ros2/rosbag2 #1880: Rpr job seems running multiple tests in parallel

MichaelOrlov, 2024-12-10:
## Description
Rpr and Jpr jobs seems running multiple tests in parallel

### Expected Behavior
Rpr and Jpr jobs run tests sequentially

### Actual Behavior
Reference to the failing Rpr job https://build.ros2.org/job/Rpr__rosbag2__ubuntu_noble_amd64/477/testReport/junit/(root)/projectroot/test_record_all__rmw_fastrtps_cpp/
From the log I see that in the `published_messages_from_multiple_topics_are_recorded` test, we've got more messages that we expected.
https://github.com/ros2/rosbag2/blob/8f1527e56bcbe13dbc5efa1f250f10471b19429a/rosbag2_transport/test/rosbag2_transport/test_record_all.cpp#L38-L55
And I see that the Rosbag2 recorder subscribed to the 
`[INFO] [1733734416.350351887] [rosbag2_recorder]: Subscribed to topic '/composable_recorder_test_string_topic'`
However, we publish on this topic only from the other test `recorder_can_automatically_start_recording_after_composition`
https://github.com/ros2/rosbag2/blob/8f1527e56bcbe13dbc5efa1f250f10471b19429a/rosbag2_transport/test/rosbag2_transport/test_composable_recorder.cpp#L268-L276

## To Reproduce
N/A
** Steps to reproduce the behavior, e.g.
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. Logs print error '...' **

## System (please complete the following information)
 - OS: Ubuntu 22.04, 24.04
 - ROS 2 Distro: Rolling, Jazzy
 - Install Method: CI, Rpr job
 - Version: Rolling

## Additional context
** Add any other context about the problem here **

MichaelOrlov, 2024-12-10:
@claraberendsen @clalancette @cottsay This is an example of the Rpr job failure that I mentioned on today's ROS PMC meeting with suspicion that the Rpr job runs multiple tests at a time.

clalancette, 2024-12-18:
> @claraberendsen @clalancette @cottsay This is an example of the Rpr job failure that I mentioned on today's ROS PMC meeting with suspicion that the Rpr job runs multiple tests at a time.

Based on the logs, I don't think that it is running tests in parallel.  First of all, within a package we never run tests in parallel; they are always run sequentially by `ctest`.  However, there is some possibility that colcon is running multiple packages in parallel.  But if you look in the logs, you see this:

```
Invoking '. /opt/ros/rolling/setup.sh && PYTHONIOENCODING=utf_8 PYTHONUNBUFFERED=1 colcon test --build-base build_isolated --install-base install_isolated --test-result-base test_results --event-handlers console_direct+ --executor sequential --test-result-base /tmp/ws/test_results' in '/tmp/ws'
```

Which is very clearly using the sequential executor.  That means it will not run packages in parallel, but one-at-a-time.

My guess is that something else is going on here; maybe a previous test in the same package isn't properly cleaning up after itself.

MichaelOrlov, 2025-03-01:
- Closing this issue for now, as I believe tests are not running in parallel. For root cause of the extra messages please refer to the https://github.com/ros2/rosbag2/issues/1914#issuecomment-2669973810

closed 2025-03-01 by MichaelOrlov