ros2/rosbag2 #1891: `--start-offset` as non-negative float (relax equality to allow 0 as input)

23pointsNorth, 2024-12-30:
## Description

Currently, based on https://github.com/ros2/rosbag2/pull/931, the input cli param `--start-offset` has a default value of `0.0`, yet is of type [`check_positive_float` ](https://github.com/ros2/rosbag2/blob/rolling/ros2bag/ros2bag/api/__init__.py#L118).

This creates the issue, that by default, the value is 0, yet one cannot set the value to 0 by not using hte default field. i.e. 

```
ros2 bag run {bag_name} --start-offset 0 # this fails
```

```
ros2 bag run {bag_name}  # this works as expected, yet starts from 0
```

## Completion Criteria

It would be great to relax the equality, as it would allow using the start-offset param as part of matrix scans in CI jobs and other places where a list of offsets may be chosen.

P.S. `check_positive_float` and [`positive_float`](https://github.com/ros2/rosbag2/blob/7cd158f6163cc7f2538140a1528645669c55f641/ros2bag/ros2bag/verb/play.py#L31) have been used mixed within the argparse.

AntoineHX, 2025-01-29:
I tried the command that was supposed to fail with the recording of the talker demo node. I didn't noticed any issue on the default branch (default). Maybe your issue is on another branch ?

MichaelOrlov, 2025-01-31:
@23pointsNorth @AntoineHX I am curious if the issue you are reporting here has already been fixed in the https://github.com/ros2/rosbag2/pull/1682/files ?

23pointsNorth, 2025-01-31:
It looks like it has been. Just have to check if the version of humble in apt has been backported.

MichaelOrlov, 2025-02-01:
- Closing as already fixed in the https://github.com/ros2/rosbag2/pull/1682

closed 2025-02-01 by MichaelOrlov