ros2/rosbag2 #1892: New `ros2 bag echo` verb to dump a single topic to stdout?

moooeeeep, 2025-01-07:
## Description

I was looking to dump a single topic's content to stdout in csv format, like `rostopic echo -b <bagfile> -p <topic>` used to do.

Related question: https://robotics.stackexchange.com/questions/114194/exporting-single-topic-data-to-csv-with-rosbag2-bag-files 

I had the impression that this feature doesn't exist, so I wrote a small script using the Python API. I wondered if it would be useful to add this as a new verb to the cli (the actual name would be open for debate)?

## Related Issues

None

## Completion Criteria

* CLI verb to dump topic from bagfile to stdout (yaml and/or csv format)
* Required customization options, e.g., to specify storage id and output format

## Implementation Notes / Suggestions

Similar to the [test code of the `SequentialReader`](https://github.com/ros2/rosbag2/blob/rolling/rosbag2_py/test/test_sequential_reader.py#L38), but with formatted output.

## Testing Notes / Suggestions

Process known bagfiles for different sets of options and verify the captured output.

MichaelOrlov, 2025-01-10:
Hi @moooeeeep I would imagine that this functionality would be an extension/addition to the `ros2 bag convert` CLI command.
Please refer to the https://github.com/ros2/rosbag2/blob/rolling/README.md#converting-bags for details about convert verb usage and possible options.

omsv135, 2025-08-22:
Hi,
I would like to look into this enhancement, however, I am not sure what the tag "backlog" implies. Is it worth working on?

MichaelOrlov, 2025-09-02:
Hi @omsv135, Yes. If you would like to help with these enhancements, your help would be appreciated.

However, please be aware of my comment above
> I would imagine that this functionality would be an extension/addition to the ros2 bag convert CLI command.
Please refer to the https://github.com/ros2/rosbag2/blob/rolling/README.md#converting-bags for details about convert verb usage and possible options.

omsv135, 2025-09-02:
Definitely, I will add the functionality to `ros2 bag convert` CLI command.

moooeeeep, 2025-09-09:
FWIW this is the code we ended up using: https://gist.github.com/moooeeeep/061c9d4d35475364af09f70e8ac2c174

For my understanding: Integrating this into the convert verb would involve deriving some new `FormattedStdoutWriter` implementation from `SequentialWriter`, to which the `ReaderWriterFactory::make_writer()` would then delegate, based on some to-be-added `RecordOptions` ?

MichaelOrlov, 2025-09-15:
> FWIW this is the code we ended up using: https://gist.github.com/moooeeeep/061c9d4d35475364af09f70e8ac2c174
>
> For my understanding: Integrating this into the convert verb would involve deriving some [...]

That reminds me about https://github.com/ros2/rosbag2/blob/rolling/rosbag2_examples/rosbag2_examples_py/rosbag2_examples_py/rosbag2csv.py, which I wrote 5 months ago or so.

In my opinion, the script is pretty simple and trivial and would be pretty easy to modify to save or output data in text format or whatever is needed in Python. 

I am curious if developing a special converter built into `ros2 bag convert` CLI still makes sense?

cc: @omsv135

omsv135, 2025-09-20:
> FWIW this is the code we ended up using: https://gist.github.com/moooeeeep/061c9d4d35475364af09f70e8ac2c174

Thanks for the code. I think some effort could have been saved using `rosidl_runtime_py.convert.message_to_yaml` and `rosidl_runtime_py.convert.message_to_csv`. I saw these being used in the example that @MichaelOrlov provided.

> I am curious if developing a special converter built into `ros2 bag convert` CLI still makes sense?

While processing a single bag is trivial, it would be useful for getting messages from multiple bags. Though, even that could be achieved by first merging the bags then using the example script.

I think we should develop this if combining topics from multiple bags into a csv file is a common task for many users, which may be true.