ros2/rosbag2 #1898: Project Logo

SteveMacenski, 2025-01-17:
Hi,

I was working on some slides for rosbag2 educational material for a client and wanted to put together a graphic for introducing the software. I noticed rosbag2 doesn't have a logo yet, so I played around with DALLE for a few minutes and generated something I thought was pretty good:

[image]

If the maintainers here are of a similar opinion, I'd be happy to remove the background, clean up those edges around the top of the bag where it meets the square, and submit the image files for this project to use as a logo

If not, no worries. I just figured I'd offer. Its always cool when projects have their own logos :-)

fujitatomoya, 2025-01-18:
I gotta say I will leave this to @SteveMacenski @MichaelOrlov @christophebedard ... i do not have good taste for this kind of design... 😅

MichaelOrlov, 2025-01-18:
@SteveMacenski What should be the size of the icon in pixels?

MichaelOrlov, 2025-01-18:
Honestly, I would imagine the icon a bit different.  
Instead of putting "ROS" dots logo inside bag it would be more informative to put set of "documnets" aka messages in a big bag. 
The surrounding square is an extra visual load and is not needed.
The scale underneath is a good idea. However, the proposed form is a bit misleading, with random red dots up and down. We need something more visually recognisable as a time scale.

SteveMacenski, 2025-01-18:
> What should be the size of the icon in pixels?

More or less what this image is, I can upsample it / downsample it later if different sizes are necessary, but I'd start here, its plenty big for a readme / github org icon. If someone in the future liked it enough to want to blow it up large for stickers, I could find a way to do that. 

> Honestly, I would imagine the icon a bit different.

The riffs I can do easily:
* Take only the "rosbag" part of the icon (throw away the background and the square with the red dots and time scale)
* Take the full icon, remove the background, fix up the top of the bag meeting the square boundaries, making the "red dots" on the scale actual circles and in more reasonable places, and probably removing the random 3 dots on the upper right corner

Anything else that are major modifications are not things I can do with this given that this isn't a normal illustrator file I can really mess with things regarding (its a DALLE GenAI output like I mentioned). If none of those options entice you, I totally understand 😄 I would be particular about the logo/icon for my project too. This is just the best I can do without literally going out and paying a designer on Fiverr. 

>  Instead of putting "ROS" dots logo inside bag it would be more informative to put set of "documnets" aka messages in a big bag.

This is where you lose what I can do here. Even if I had raw professional illustrator files, I'm shit with font selection and I'd make it look awful if there's was any text. 

> However, the proposed form is a bit misleading, with random red dots up and down. 

I think abstract things are fine IMO. Nav2's logo is a hot air balloon - people are just happy to have something cool. I wouldn't over think it trying to make it too literal if it look cool and it gets the general gist across. But that's just my opinion which has no weight on your project 😆

TKaluza, 2025-01-18:
So maybe I can provide some work for the logo, doing a clean handmade svg, if it's feasible for me ;)

> Honestly, I would imagine the icon a bit different. Instead of putting "ROS" dots logo inside bag it would be more informative to put set of "documnets" aka messages in a big bag. The surrounding [...]

yes, I agree. 

Is the bag a fixed cloth bag, like an old money bag? 

Or could it also be something like this, i do like the bag as a "bowl" of bits and bytes. 
[image]

Or should it go in proposed direction (the look of the bag - not everything else)
[image]

MichaelOrlov, 2025-01-18:
@SteveMacenski @TKaluza Something like this
[image]

TKaluza, 2025-02-01:
[image]

What about that?