ros2/rosbag2 #1918: Sequentially play a directory or list of bagfiles without metadata.yaml

emersonknapp, 2025-02-22:
## Description

With continuous or just long recording sessions, users need to retrieve and play back just a specific subset of files or a specific directory, which will not have `metadata.yaml`. That data should not be required to sequentially play a series of files.

## Related Issues

** Is this feature dependent on any other features? Is it part of a larger project? Note here. **

## Completion Criteria

- `ros2 bag play directory_without_metadata/` will play all discovered files in filesystem order (alphabetical)
- `ros2 bag play fileA.mcap fileB.mcap fileC.mcap` will play all specified files sequentially in the order specified

Maybe the directory option isn't required, because the user could do file globbing, 

```bash
ros2 bag play my_recording/*.mcap
```

Which would be fine except you lose autodetection, and if you do 

```bash
ros2 bag play my_recording/*
```

then it might choke on for example `metadata.yaml`, which isn't a bagfile.

## Implementation Notes / Suggestions

I believe this comes down to removing the `metadata.yaml` requirement checks from `SequentialReader` - and instead falling back to just reading the directory.

The multifile option could be as simple as `ros2bag` CLI allowing `nargs='+'`, then running `for bag in args.bags: Play()` though this would be a naive implementation that would lose Publisher continuity and introduce latency on file boundaries. So probably better to pass the list of files to `SequentialReader` which would override the `metadata.yaml` check.

## Testing Notes / Suggestions

** All features in this project need tests. Please give some input on cases that will need to be tested - and how the testing might be implemented. **

MichaelOrlov, 2025-02-25:
@emersonknapp I was thinking about what if add some intelligence to the Rosbag2 player. In case if someone want to replay the directory with files without `metadata.yaml` file ( i.e. `ros2 bag play directory_without_metadata/`) the player would do automatically reindexing and creating metadata on the fly the same way as `ros2 bag reindex directory_without_metadata/` does but without saving metadata.yaml file?
https://github.com/ros2/rosbag2/blob/0105eec1d0bf650280b32076234fee73236bca31/rosbag2_cpp/src/rosbag2_cpp/reindexer.cpp#L250-L275

It seems we already have all the necessary heavy-lifting code in the reindexer class - and the fix is going to be literally in 10 lines of code or so. The benefit of it is a consistent behavior with the `ros2 bag reindex` CLI option.
We will need to rewrite this part of "`else`" code
https://github.com/ros2/rosbag2/blob/0105eec1d0bf650280b32076234fee73236bca31/rosbag2_cpp/src/rosbag2_cpp/readers/sequential_reader.cpp#L112-L131
to use reindexing instead of reading metadata from bag file directly.

Thoughts?

emersonknapp, 2025-02-25:
Yeah, I think that a "reindex on-demand" type of implementation makes sense to me! It would then validate right at the beginning that the files are playable. I think we'd want to allow for it skipping files it can't play, like other meta-information files, rather than failing, but that's an error handling detail.

MichaelOrlov, 2025-02-25:
Currently, we have a specific regex for choosing bag file names with recording. Log files and other specific non-robag2 files should already be filtered out by default.