ros2/rosbag2 #1920: How to play compressed rosbag?

scrouthtv, 2025-02-22:
## Description
I did record a rosbag with this configuration:
```bash
cat mcap_config.yaml
compresion: "Zstd"
ros2 bag record -s mcap --all --storage-config-file mcap_config.yaml
```

The bag is very long and contains, among others, lidar data, resulting in 400 GB of data.
When trying to play this bag, I only get a single line of output:
```
[rosbag2_compression]: Decompressing [...]_0.mcap.zstd
```
but no topics are published.

### Expected Behavior
The bag's contents is published.

### Actual Behavior
Only this output, I guess the entire bag is decompressed **before** playing it back (?)
I am wondering whether it is possible to play a compressed back without decompressing the entire bag first? If not, doesn't that defeat the purpose of compressing the bag?

## System (please complete the following information)
 - OS: Ubuntu Jammy
 - ROS 2 Distro: Humble
 - Install Method: apt
 - Version: rosbag2_compression is version 0.15.13, I don't know if there is any other version information I can provide

MichaelOrlov, 2025-02-24:
@scrouthtv If you would use 
```
cat mcap_config.yaml
compresion: "Zstd"
ros2 bag record -s mcap --all --storage-config-file mcap_config.yaml
```
You shouldn't get a bag file with a name like [...]_0.mcap.**zstd**. It should be a regular mcap file without `zstd` extension at the end

It looks like you were using `ros2 bag record -a --compression-mode file --compression-format zstd` command instead.
- Please refer to the relevant issue and description in the https://github.com/ros2/rosbag2/issues/1533

scrouthtv, 2025-02-26:
I just looked into this again. Actually, I was using neither two ways, but `ros2 bag convert -o convert.yaml` with
```yaml
output_bags:
- uri: out
  storage_id: mcap
  compression_mode: file
  compression_format: zstd
  all_topics: true
```

This is giving bags with `[...]_0.mcap.zstd`, I just tested again.

MichaelOrlov, 2025-03-01:
@scrouthtv You've got what you were asking for in your conversion operation.
However, it seems not what you want.
You likely need to use 
```
  storage-config-file: mcap_config.yaml
```
instead of 
```
  compression_mode: file
  compression_format: zstd
```

scrouthtv, 2025-03-06:
Ah okay. Re- converting the rosbag with your parameters creates a playable rosbag. This is configuration option is however not documented in the [rosbag2 readme](https://github.com/ros2/rosbag2?tab=readme-ov-file#converting-bags).

closed 2025-03-06 by scrouthtv