ros2/rosbag2 #1922: python3 new_data_colletion_test.py  [ERROR] [1740439581.729711754] [rosbag2_storage_mcap]: Message type name 'RealsenseMetadata' is not understood by type definition search, definition wll be left empty.

Akumar201, 2025-02-24:
I am new to rosbag , I am not sure why I am getting this error 

python3 new_data_colletion_test.py  [ERROR] [1740439581.729711754] [rosbag2_storage_mcap]: Message type name 'RealsenseMetadata' is not understood by type definition search, definition wll be left empty.
```

import rclpy
from rclpy.node import Node
from rclpy.serialization import serialize_message
from std_msgs.msg import String
from realsense2_camera_msgs.msg import Metadata as RealsenseMetadata

```
```
import rosbag2_py

class SimpleBagRecorder(Node):
    def __init__(self):
        super().__init__('simple_bag_recorder')
        self.writer = rosbag2_py.SequentialWriter()

        storage_options = rosbag2_py.StorageOptions(
            uri='/home/max/my_bag2',
            storage_id='mcap')
        converter_options = rosbag2_py.ConverterOptions('', '')
        self.writer.open(storage_options, converter_options)

        topic_info = rosbag2_py.TopicMetadata(
            name='/cam_right_wrist/D405_right_wrist/depth/metadata',
            type='RealsenseMetadata',
            serialization_format='cdr')
        self.writer.create_topic(topic_info)

        self.subscription = self.create_subscription(
            String,
            '/cam_right_wrist/D405_right_wrist/depth/metadata',
            self.topic_callback,
            10)
        self.subscription

    def topic_callback(self, msg):
        self.writer.write(
            'chatter',
            serialize_message(msg),
            self.get_clock().now().nanoseconds)

def main(args=None):
    rclpy.init(args=args)
    sbr = SimpleBagRecorder()
    rclpy.spin(sbr)
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

MichaelOrlov, 2025-03-02:
Hi @Akumar201  
- You see those warnings because the Rosbag2 recorder tried to search for the message definition in your local setup and failed to find any for the `RealsenseMetadata' type name. 
- That is not a fatal error, and the following playback of such a bag file should be Okay. However, some introspection tools, like Foxglove Studio, may rely on the saved message definitions in some cases.

Akumar201, 2025-03-04:
Thank You @MichaelOrlov , I will close this issue now.

closed 2025-03-04 by Akumar201