ros2/rosbag2 #1925: Calling rosbag2_cpp::Reader::open from multiple threads leads to crashes

MaxFleur, 2025-02-27:
## Description
I am writing an application where users can edit bag files to create a new, edited bag file. This includes a feature to rename topics in the original bag file.\
To rename topics, a `rosbag2_cpp::Reader` instance is used. This will open the original bag file, extract its metadata and rename the topic inside it. An additional instance of `rosbag2_cpp::Writer` then creates the renamed topic and later writes read messages into the new, edited bag file.\
To speed things up, I've parallelized the algorithm to use a separate thread for each original bag topic. So, for example, if the original bag has four topics, four threads are used.\
However, I realized that if the function `rosbag2_cpp::Reader::open` is called in multiple threads simultaneously, this will lead to crashes. Enclosing this code in a mutex will resolve the problem, though.\
I've included a minimal project as well as an example bag file for reproducing the error. This minimum project just opens the bag file in multiple threads simultaneously.

**Note:** I am well aware that it might be unusual to call `rosbag2_cpp::Reader::open` multiple times simultaneously and that it's not a bug, but rather intended to be called sequentially. But I do not see a reason why it should not be possible to open this bag file multiple times for read only access.

### Expected Behavior
The bag file is opened and read by different threads at once. 

### Actual Behavior
Sometimes the executed function leads to crashes.

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Copy the provided "thread_test" dir to your ros2_workspace/src.
2. Copy the provided bagfile to an arbitrary location.
3. Adjust line 11 in `main.cpp` to include the bag file location.
4. `cd path/to/ros2_workspace`
5. `colcon build`
6. `source install/setup.bash`
7. `ros2 run thread_test thread_test`
8. To reproduce the error, comment out line 19 and 21 in `main.cpp`.

## System (please complete the following information)
 - Ubuntu 22.04
 - ROS 2 Distro: Humble
 - Install Method: APT
 - Version: 0.10.0-1jammy.20250211.160146

## Additional context

Backtrace:
```
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737310099008) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737310099008) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737310099008, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff6c42476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff6c287f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff70b042a in __gnu_cxx::__verbose_terminate_handler() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff70ae20c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff70ae277 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff70ae4d8 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007ffff7f55100 in ?? () from /opt/ros/humble/lib/librosbag2_cpp.so
#10 0x00007ffff7f683e5 in rosbag2_cpp::readers::SequentialReader::open(rosbag2_storage::StorageOptions const&, rosbag2_cpp::ConverterOptions const&) () from /opt/ros/humble/lib/librosbag2_cpp.so
#11 0x00007ffff7f65604 in rosbag2_cpp::Reader::open(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/humble/lib/librosbag2_cpp.so
#12 0x0000555555557b81 in main::$_1::operator()() const ()
#13 0x0000555555557af9 in main::$_0::operator()() const ()
#14 0x0000555555557ad5 in void std::__invoke_impl<void, main::$_0>(std::__invoke_other, main::$_0&&) ()
#15 0x0000555555557a95 in std::__invoke_result<main::$_0>::type std::__invoke<main::$_0>(main::$_0&&) ()
#16 0x0000555555557a6d in void std::thread::_Invoker<std::tuple<main::$_0> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) ()
#17 0x0000555555557a45 in std::thread::_Invoker<std::tuple<main::$_0> >::operator()() ()
#18 0x0000555555557999 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<main::$_0> > >::_M_run() ()
#19 0x00007ffff70dc253 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#20 0x00007ffff6c94ac3 in start_thread (arg=<optimised out>) at ./nptl/pthread_create.c:442
#21 0x00007ffff6d26850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

[thread_test.tar.gz](https://github.com/user-attachments/files/19007789/thread_test.tar.gz)
[dummy_bag.tar.gz](https://github.com/user-attachments/files/19007779/dummy_bag.tar.gz)

MichaelOrlov, 2025-03-18:
Hi @MaxFleur,
It seems idea of reading a bag from multiple threads is ill-formed by design. 
Because:
1. The bottleneck is usually on the writer's side.
2. Reading bag and processing messages in multiple threads will cause changes in the order of the messages in the rewritten bag. Please note that we have the assumption that messages are ordered in the bag file by the received timestamp. One will need to make synchronization on the writer side to make sure that messages in the correct order.

The more correct design would be to have a cache buffer with one writer and multiple readers where you can periodically read a bulk of messages and process them as you want.

MaxFleur, 2025-03-18:
@MichaelOrlov thanks for some clarification, although I do not really see why just reading and not writing should be ill-formed:
1. The bottleneck has been taken care of due to writing with multiple writers at once.
2. While reading and processing messages in multiple threads my cause changes in the message order, in this case this shouldn't be a problem - for each selected topic which will be in the edited, to-be-created-bagfile, we have its own, separate writer, which will then write messages for just this one topic in sequential order. So the synchronization is there by default.

All this does not explain why `reader.open()` crashes for multiple threads at once, though - in my understanding, this should not be the cause, because no data should be manipulated just by reading something.

MichaelOrlov, 2025-03-18:
@MaxFleur Regardless of the upper level design choice. The Rosbag2 reader class wasn't designed to operate in the multi-threaded environment. In your particular case the crash happened because Rosbag2 was instructed to use SQLite3 storage plugin. 
In the [sqlite3 storage plugin](https://github.com/ros2/rosbag2/blob/893c73150118e5fd2f0505c2af818c1946d02e3d/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_wrapper.cpp#L43-L45) we are opening database with the `SQLITE_OPEN_NOMUTEX`
```cpp
if (io_flag == rosbag2_storage::storage_interfaces::IOFlag::READ_ONLY) {
    int rc = sqlite3_open_v2(
      uri.c_str(), &db_ptr,
      SQLITE_OPEN_READONLY | SQLITE_OPEN_NOMUTEX, nullptr);
```
We do this on purpose for the sake of performance.
From ChatGPT explanation:
> SQLITE_OPEN_NOMUTEX is a flag you can provide to the SQLite sqlite3_open_v2() function (or related functions) that tells SQLite not to use internal mutexes (locks) for that database connection. In other words, it disables SQLite’s internal thread-safety mechanisms for this particular connection.

In other words sqlite3 storage plugin doesn't support simultaneous access to the database from multiple threads.

MichaelOrlov, 2025-03-19:
Also provided example fails to load multiple instances of the Rosbag2 storage plugin because of the outdated version of the class loader on Humble.

- Need to backport a relevant fixes from the https://github.com/ros/class_loader/pull/201 and https://github.com/ros/class_loader/pull/205. But this is out of scope of the Rosbag2 and Tooling WG responsibility.

MaxFleur, 2025-03-19:
I understand. So it is effectively a won't-fix? In this case, I guess the issue can be closed.

Maybe it is a good idea to specify somewhere in the documentation that reading and writing in multi-threaded setups is not supported or recommended because it might lead to crashes or inconsistency.

MichaelOrlov, 2025-03-21:
- @MaxFleur Thank you for understanding. Closing as wouldn't be fixed.

closed 2025-03-21 by MichaelOrlov