ros2/rosbag2 #1940: Memory leak from saving IMU messages from Jazzy in Humble

EmmanuelMess, 2025-03-16:
For the simulation I have a Harmonic container based on `ghcr.io/sloretz/ros:jazzy-simulation` and I run `gz sim -r` with an sdf with:

```xml
      <link name="imu_link">
        <pose>0 0 0 0 0 0</pose>
        <sensor name='imu_gnss' type='imu'>
          <topic>gnss/imu</topic>
          <update_rate>200</update_rate>
          <always_on>true</always_on>
          <visualize>true</visualize>
        </sensor>
      </link>
```

Then do:

```bash
ros2 run ros_gz_bridge parameter_bridge /clock@rosgraph_msgs/msg/Clock[gz.msgs.Clock &
ros2 run ros_gz_bridge parameter_bridge /gnss/imu@sensor_msgs/msg/Imu[gz.msgs.IMU &
```

for recording a bag I do:

```bash
docker network create -d bridge gazebo-network
docker network connect gazebo-network gnss_sim_container # Connect the other container
docker run -it --rm --network gazebo-network \
           --volume ${PWD}/output:/output:rw \
           --workdir /output \
           osrf/ros:humble-desktop-full /bin/bash
           
cd ~/ros2_ws
colcon build # Builds some custom messages
source install/setup.bash
cd /output
ros2 bag record --use-sim-time --storage sqlite3 --all
```

After stopping the recording with ^C it eats up 20GB of RAM *after* it prints "Recording stopped". If I don't have the IMU it doesn't leak.

MichaelOrlov, 2025-03-18:
The memory leak is unlikely related to the Rosbag2 because:
1. It reproduces only with the specific message data type (sensor_msgs/msg/Imu) and node configuration like `ros_gz_bridge parameter_bridge.` The Rosbag2 recorder deals only with the deserialized messages in the default configuration and is agnostic to message specifics.
2. We do not observe this memory leak in our environment.

There are too many components involved in the reproduction scenario that could cause the memory leak, e.g., the gazebo-network bridge, the gnss_sim_container and a mix of ROS 2 distros (Humble and Jazzy). 
Please provide a minimum valuable reproducer with the minimum SW components involved.

EmmanuelMess, 2025-03-18:
I have removed the gnss sim component, and I'm testing a simple example for you. For now, maybe this image indicates where the issue.

[image]

EmmanuelMess, 2025-03-18:
Replication:

Console 1:
```
python3 -m venv venv
source venv/bin/activate
python3 -m pip install rocker

rocker --name oom_container --volume ${PWD}/imu.sdf:/imu.sdf:ro --x11 ghcr.io/sloretz/ros:jazzy-simulation

gz sim imu.sdf &
```

Console 2:
```
docker network create -d bridge gazebo-network
docker network connect gazebo-network oom_container
docker run -it --rm --network gazebo-network \
           --workdir /output \
           osrf/ros:humble-desktop-full /bin/bash
           
ros2 bag record --use-sim-time --storage sqlite3 --all

```

Console 1:
```
ros2 run ros_gz_bridge parameter_bridge /clock@rosgraph_msgs/msg/Clock[gz.msgs.Clock &
ros2 run ros_gz_bridge parameter_bridge /gnss/imu@sensor_msgs/msg/Imu[gz.msgs.IMU &

ros2 topic echo /gnss/imu
```

Start simulation

emersonknapp, 2025-03-21:
The first thing I notice is that this test is doing cross-distro communication. Your IMU publisher from the gz bridge is running in ROS 2 Jazzy, and the recorder is running in ROS 2 Humble.

While I'm not sure what effect this has (it's not officially supported though in many cases it works), could you try running your recorder in Jazzy also and see if you get the same results?

EmmanuelMess, 2025-03-21:
> While I'm not sure what effect this has (it's not officially supported though in many cases it works), could you try running your recorder in Jazzy also and see if you get the same results?

With both the sender and receiver in jazzy I don't see the memory leak.

closed 2025-03-22 by emersonknapp