ros2/rosbag2 #1942: rosbag2_transport::Recorder node does not record topics named relative to that node's namespace.

rodericktaylor, 2025-03-20:
## Description
rosbag2_transport::Recorder doesn't record topics if they named relative to the recorder node's namespace.

### Expected Behavior
If a recorder is launched in a namespace, the topics named relative to that namespace are recorded.

### Actual Behavior
If a recorder is launched in a namespace, the topics named relative to that namespace are not recorded.

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Launch a recorder node wihtin a namespace, specify topics to record with names relative to that namespace.
2. publish to those topics.  
3. See that the topics are not recorded.
## System (please complete the following information)
 - OS: Unbuntu noble.
 - ROS 2 Distro: Jazzy
 - Install Method: apt.
 - Version: git branch jazzy. [57922f4](https://github.com/ros2/rosbag2/commit/57922f4f3d1b657286e95e4465cd0bd2eb2ff500)

closed 2025-03-24 by MichaelOrlov

rodericktaylor, 2025-04-17:
@MichaelOrlov What is the process to generate an official release that will include this fix?

MichaelOrlov, 2025-04-21:
Will need to make a release for Jazzy. 

@marcoag It seems we have not done releases for Rosbag2 Jazzy since December last year.  
Wouldn't you mind making a Rosbag2 release when doing Jazy sync next time, or should I do it?

marcoag, 2025-04-22:
> Will need to make a release for Jazzy.
> 
> [@marcoag](https://github.com/marcoag) It seems we have not done releases for Rosbag2 Jazzy since December last year. Wouldn't you mind making a Rosbag2 release when doing Jazy sync next time, or should I do it?

Yes, I have a sync on the plans, still waiting to see if I get an answer to solve the one last regression, I'll give it one more day, otherwise I'll just go ahead with it.