ros2/rosbag2 #1945: `rosbag2_transport::bag_rewrite` with compression mode `message` leads to bag file with wrong message count

MaxFleur, 2025-03-21:
## Description
I am trying to use the `ros2bag_transport` API to compress a bag file. For this, basically the following code is used:
```
rosbag2_storage::StorageOptions inputStorage;
// Change this to wherever your bag file is
inputStorage.uri = "/path/to/input";

std::vector<rosbag2_storage::StorageOptions> inputBags;
inputBags.push_back(inputStorage);

rosbag2_storage::StorageOptions outputStorage;
outputStorage.uri = "/path/to/target";
outputStorage.storage_id = "sqlite3";

rosbag2_transport::RecordOptions outputRecord;
outputRecord.all = true;
outputRecord.compression_mode = "message";
outputRecord.compression_format = "zstd";

std::vector<std::pair<rosbag2_storage::StorageOptions, rosbag2_transport::RecordOptions> > outputBags;
outputBags.push_back({ outputStorage, outputRecord });

rosbag2_transport::bag_rewrite(inputBags, outputBags);
```
The code used to setup and call the `bag_rewrite` function is very similar to the one used in the `rosbag2_transport` tests (see [here](https://github.com/ros2/rosbag2/blob/humble/rosbag2_transport/test/rosbag2_transport/test_rewrite.cpp#L198).)

If the compression mode is set to `file`, the target bag is compressed successfully. However, if compression mode `message` is used, the target bag file has a different message count than the input bag file. Moreover, the target bag message count is different for every execution, almost random.\
These problems could be reproduced using Humble __and__ Jazzy.

### Expected Behavior
The compressed bag file has the same message count as the provided input bag file.

### Actual Behavior
The compressed bag file has a different message count than the input bag file, which is randomly changed whenever the provided code is executed.

## To Reproduce

1. Copy the provided "compression_test" dir to your ros2_workspace/src.
2. Copy the provided bagfile to an arbitrary location.
3. Adjust line 10 in main.cpp to include the bag file location.
4. `cd path/to/ros2_workspace`
5. `colcon build`
6. `source install/setup.bash`
7. `ros2 run compression_test compression_test`
8. To reproduce the error with Jazzy, uncomment line 21 and comment line 22.

## System (please complete the following information)
 - OS: Ubuntu 22.04/24.04
 - ROS 2 Distro: Humble, Jazzy
 - Install Method: APT
 - Version: 0.10.0-1jammy.20250211.160146 (humble), ros2-jazzy-20241223 (Jazzy)

## Additional context

I don't want to exclude that maybe an important parameter in the record options is missing. I didn't find this one, though.\
Maybe it is a good idea to provide one or two examples on how to use this API in the example section.

[dummy_bag.tar.gz](https://github.com/user-attachments/files/19384161/dummy_bag.tar.gz)
[compression_test.tar.gz](https://github.com/user-attachments/files/19384162/compression_test.tar.gz)

MichaelOrlov, 2025-03-21:
@MaxFleur The messages drop happened likely because you are not intilizing `options.compression_queue_size = 0;` explicitly.
Please refer to the https://github.com/ros2/rosbag2/blob/57922f4f3d1b657286e95e4465cd0bd2eb2ff500/rosbag2_py/src/rosbag2_py/_transport.cpp#L473-L480 as an example

MichaelOrlov, 2025-03-21:
By design we are dropping messages if compressor threads can't keep up with writer speed. Which is make sens during recording.  
Please refer to the relevant peace of code here https://github.com/ros2/rosbag2/blob/57922f4f3d1b657286e95e4465cd0bd2eb2ff500/rosbag2_compression/src/rosbag2_compression/sequential_compression_writer.cpp#L425-L431

emersonknapp, 2025-03-22:
If that's the issue, perhaps bag rewrite should set those values since it is a burst process.

MaxFleur, 2025-03-22:
@MichaelOrlov 
That did indeed the trick. Using `options.compression_queue_size = 0;` works as intended. Thanks for the help!

@emersonknapp
Agreed, imo a functionality to rewrite a bag file shouldn't drop any messages at all. Setting the value is a good idea here.

Maybe it is also a good idea to provide an example for `rosbag2_examples` regarding the `bag_rewrite` functionality? I could provide such an example.

MichaelOrlov, 2025-03-22:
@emersonknapp @MaxFleur We do have explicitly init `options.compression_queue_size = 0` for bag rewrite API on Python binding layer
https://github.com/ros2/rosbag2/blob/57922f4f3d1b657286e95e4465cd0bd2eb2ff500/rosbag2_py/src/rosbag2_py/_transport.cpp#L483-L508
the CLI command and Python API are immune from this issue. 
To clarify, by current design, the default initialization for `options.compression_queue_size = 0` could be overwritten by explicitly mentioning this parameter in the config files for storage options for bag rewrite. This is consistent with other parameters and expected to be this way by design.

If we override this parameter on the downstream `rosbag2_transport` layer, it could be considered a bug. When the user specifies this parameter explicitly in config files, it is overridden internally and doesn't take it into account.

We can consider changing the default value for this parameter to **0** instead of **1**, but we need to double-check if it will affect other flows and use cases.

@MaxFleur As usual, help with documentation and examples is welcome and would be appreciated.

MaxFleur, 2025-03-28:
@MichaelOrlov 
I see. So it is either a possible message dropping or overridden config file. Either way, it should be documented somewhere for better clarification.
I'd provide one or two examples for writing compressed bag files using `rosbag2_transport::bag_rewrite`. Should I open another issue for this or just assign this issue to myself?

MichaelOrlov, 2025-05-07:
@MaxFleur Please be notified that we changed the default value for the `RecordOptions.compression_queue_size` to **0** to resolve a similar issue in the https://github.com/ros2/rosbag2/pull/1975

MaxFleur, 2025-05-08:
@MichaelOrlov I understand.
Have to adjust #1956 then and specify that for Humble, this option still has to be set.

MichaelOrlov, 2025-05-12:
- @MaxFleur The fix has been backported to Humble in the https://github.com/ros2/rosbag2/pull/2012

MaxFleur, 2025-05-13:
@MichaelOrlov 
Really nice, thanks for the info! I guess we can close this issue then?

closed 2025-05-13 by MichaelOrlov