ros2/rosbag2 #1947: [Test Failure] test_record_qos_profiles doesn't work on Windows 10

YuanYuYuan, 2025-03-24:
## To Reproduce

```bash
colcon test --merge-install --event-handlers console_cohesion+ console_direct+  --packages-select ros2bag --pytest-args -k test_record_qos_profiles
```

Error log

```
Starting >>> ros2bag
============================= test session starts =============================
platform win32 -- Python 3.12.3, pytest-7.4.4, pluggy-1.4.0
cachedir: C:\dev\rolling\build\ros2bag\.pytest_cache
rootdir: C:\dev\rolling\src\ros2\rosbag2\ros2bag
configfile: pytest.ini
plugins: ament-copyright-0.19.1, ament-flake8-0.19.1, ament-lint-0.19.1, ament-mypy-0.19.1, ament-pep257-0.19.1, ament-xmllint-0.19.1, launch-pytest-3.8.0, launch-testing-3.8
.0, launch-testing-ros-0.28.0, colcon-core-0.17.1, cov-4.1.0, mock-3.12.0, repeat-0.9.3, rerunfailures-12.0, timeout-2.2.0
collected 32 items / 31 deselected / 1 selected

test\test_record_qos_profiles.py F                                       [100%]

================================== FAILURES ===================================
___________________ launch tests: test_record_qos_profiles ____________________

---------------------------- Captured stdout call -----------------------------
[INFO] [launch]: All log files can be found below C:\Users\user\.ros\log\2025-03-24-16-10-40-127461-DESKTOP-KN94OIO-2112
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2bag-cli-1]: process started with pid [1740]
[ros2bag-cli-1] [ERROR] [ros2bag]: Time overrides must include both seconds (sec) and nanoseconds (nsec).
[WARNING] [ros2bag-cli-1]: 'SIGINT' sent to process[ros2bag-cli-1] not supported on Windows, escalating to 'SIGTERM'
[INFO] [ros2bag-cli-1]: sending signal 'SIGTERM' to process[ros2bag-cli-1]
[ERROR] [ros2bag-cli-1]: process has died [pid 1740, exit code 1, cmd 'ros2 bag record -a --qos-profile-overrides-path C:/dev/rolling/src/ros2/rosbag2/ros2bag/test/resources/
incomplete_qos_duration.yaml --output C:/Users/user/AppData/Local/Temp/tmpvzo12rjh/ros2bag_test_incomplete_duration'].
[INFO] [ros2bag-cli-2]: process started with pid [1492]
[WARNING] [ros2bag-cli-2]: 'SIGINT' sent to process[ros2bag-cli-2] not supported on Windows, escalating to 'SIGTERM'
[INFO] [ros2bag-cli-2]: sending signal 'SIGTERM' to process[ros2bag-cli-2]
[ERROR] [ros2bag-cli-2]: process has died [pid 1492, exit code 1, cmd 'ros2 bag record -a --qos-profile-overrides-path C:/dev/rolling/src/ros2/rosbag2/ros2bag/test/resources/
incomplete_qos_profile.yaml --output C:/Users/user/AppData/Local/Temp/tmpvzo12rjh/ros2bag_test_incomplete'].
[ros2bag-cli-2] stdin is not a terminal or console device. Keyboard handling disabled.[INFO] [1742803841.773830000] [rosbag2_recorder]: Press SPACE for pausing/resuming

[INFO] [ros2bag-cli-3]: process started with pid [1056]
[ros2bag-cli-3] usage: ros2 bag record [-h] [-o OUTPUT] [-s {my_test_plugin,mcap,sqlite3}]
[ros2bag-cli-3]                        [--topics Topic [Topic ...]]
[ros2bag-cli-3]                        [--services ServiceName [ServiceName ...]]
[ros2bag-cli-3]                        [--topic-types TopicType [TopicType ...]] [-a]
[ros2bag-cli-3]                        [--all-topics] [--all-services] [-e REGEX]
[ros2bag-cli-3]                        [--exclude-regex EXCLUDE_REGEX]
[ros2bag-cli-3]                        [--exclude-topic-types ExcludeTopicTypes [ExcludeTopicTypes ...]]
[ros2bag-cli-3]                        [--exclude-topics Topic [Topic ...]]
[ros2bag-cli-3]                        [--exclude-services ServiceName [ServiceName ...]]
[ros2bag-cli-3]                        [--include-unpublished-topics]
[ros2bag-cli-3]                        [--include-hidden-topics] [--no-discovery]
[ros2bag-cli-3]                        [-p POLLING_INTERVAL] [--ignore-leaf-topics]
[ros2bag-cli-3]                        [--qos-profile-overrides-path QOS_PROFILE_OVERRIDES_PATH]
[ros2bag-cli-3]                        [-f {s,a}] [-b MAX_BAG_SIZE] [-d MAX_BAG_DURATION]
[ros2bag-cli-3]                        [--max-cache-size MAX_CACHE_SIZE]
[ros2bag-cli-3]                        [--disable-keyboard-controls] [--start-paused]
[ros2bag-cli-3]                        [--use-sim-time] [--node-name NODE_NAME]
[ros2bag-cli-3]                        [--custom-data [KEY=VALUE ...]] [--snapshot-mode]
[ros2bag-cli-3]                        [--log-level {debug,info,warn,error,fatal}]
[ros2bag-cli-3]                        [--storage-config-file STORAGE_CONFIG_FILE]
[ros2bag-cli-3]                        [--storage-preset-profile {none,fastwrite,zstd_fast,zstd_small}]
[ros2bag-cli-3]                        [--compression-queue-size COMPRESSION_QUEUE_SIZE]
[ros2bag-cli-3]                        [--compression-threads COMPRESSION_THREADS]
[ros2bag-cli-3]                        [--compression-threads-priority COMPRESSION_THREADS_PRIORITY]
[ros2bag-cli-3]                        [--compression-mode {none,file,message}]
[ros2bag-cli-3]                        [--compression-format {zstd,fake_comp}]
[ros2bag-cli-3]                        [[Topic ...] ...]
[ros2bag-cli-3] ros2 bag record: error: argument --qos-profile-overrides-path: can't open 'C:/dev/rolling/src/ros2/rosbag2/ros2bag/test/resources/foobar.yaml': [Errno 2] No s
uch file or directory: 'C:/dev/rolling/src/ros2/rosbag2/ros2bag/test/resources/foobar.yaml'
[... 62 lines cut ...]
    self._rmtree(self.name, ignore_errors=self._ignore_cleanup_errors)
  File "C:\dev\.pixi\envs\default\Lib\tempfile.py", line 930, in _rmtree
    _shutil.rmtree(name, onexc=onexc)
  File "C:\dev\.pixi\envs\default\Lib\shutil.py", line 820, in rmtree
    return _rmtree_unsafe(path, onexc)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\dev\.pixi\envs\default\Lib\shutil.py", line 643, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onexc)
  File "C:\dev\.pixi\envs\default\Lib\shutil.py", line 648, in _rmtree_unsafe
    onexc(os.unlink, fullname, err)
  File "C:\dev\.pixi\envs\default\Lib\tempfile.py", line 905, in onexc
    _os.unlink(path)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\user\\AppData\\Local\\Temp\\tmpvzo12rjh\\ros2bag_te
st_basic\\ros2bag_test_basic_0.mcap'

----------------------------------------------------------------------
Ran 4 tests in 6.206s

FAILED (errors=1)
```

## System (please complete the following information)
 - OS: Windows 10
 - ROS 2 Distro: rolling (following this [guide](https://docs.ros.org/en/rolling/Installation/Alternatives/Windows-Development-Setup.html) 
 - RMW: Neither `rmw_cyclonedds_cpp` nor `rmw_zenoh_cpp` passed.

MichaelOrlov, 2025-03-24:
It seems that this test started failing for some reason after we moved to the Pixi on Windows. 
The further more deeper triaging is needed.

MichaelOrlov, 2025-03-24:
## Preliminary RCA:
- It seems with Pixi we are getting `PermissionError` exception instead of more generic `OSError` when trying to delete temporarily created file during the test's recording with `cls.tmdir.cleanup`.
As a fix, it likely will be enough to add `PermissionError` exception handler
```python
 @classmethod
    def tearDownClass(cls):
        try:
            cls.tmpdir.cleanup()
        except (OSError, PermissionError):
            if sys.platform != 'win32':
                raise
            # HACK to allow Windows to close pending file handles
            time.sleep(3)
            cls.tmpdir.cleanup()
```

closed 2025-03-25 by MichaelOrlov