ros2/rosbag2 #1953: Bag not recorded on hard shutdown or docker stop

EnzoGhisoni, 2025-03-26:
## Description
I'm running ROS2 Jazzy in a Docker Container. In our use case, the robot is stopped by unplugging the power supply (hard shutdown). The same issue occurs if the docker is stopped.

### Expected Behavior
We would like to be able to record the bag and replay it even if the system is hard shutdown (or the docker stopped).

### Actual Behavior
The mcap file generated is empty and no metadatafile is generated.

## To Reproduce
To reproduce it, publish on a topic in loop
`ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "{linear: {x: 0.1, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.0}}"`

Start the recording using the command:
`ros2 bag record -o test_bag /cmd_vel`

Stop the docker and start it again. The mcap file will be empty without metadata file.

## System (please complete the following information)
 - Docker container
 - OS: Ubuntu 24
 - ROS 2 Distro: Jazzy
 - Install Method: APT

## Additional context
It could be related to this issue: https://github.com/ros2/rosbag2/issues/901

fujitatomoya, 2025-03-26:
@EnzoGhisoni thanks for creating issue.

> In our use case, the robot is stopped by unplugging the power supply (hard shutdown). The same issue occurs if the docker is stopped.

these are not the same issue.
i can see the 2 complete different situation here, the one is software, the other is hardware.

for software, docker container situation, i do not think this is a bug in rosbag2.

1st of all, rosbag2 supports the signal handler for SIGINT and SIGTERM as following.

https://github.com/ros2/rosbag2/blob/d8f99a778183fc6ce9711a4743cb2c11e6b1756d/rosbag2_py/src/rosbag2_py/_transport.cpp#L206-L212

you are seeing this problem because you are not sending the signal to the appropriate process ID, but only main process in the container.

please see more details for https://docs.docker.com/reference/cli/docker/container/stop/

after all, it ends up sending the SIGKILL to the main process to shutdown the container, that brings down all other processes in the container.

if we send the either SIGINT or SIGTERM to the `ros2 bag record` process ID, rosbag2 cancels the recording and shutdown normally.

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 bag record -o test_bag /cmd_vel
[WARN] [ros2bag]: Positional "topics" argument deprecated. Please use optional "--topics" argument instead.
[INFO] [1743007447.840195603] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1743007447.843931532] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1743007447.844168926] [rosbag2_recorder]: Listening for topics...
[INFO] [1743007447.844344150] [rosbag2_recorder]: Recording...
[INFO] [1743007448.048198140] [rosbag2_recorder]: Subscribed to topic '/cmd_vel'
[INFO] [1743007448.148319951] [rosbag2_recorder]: All requested topics are subscribed. Stopping discovery...
[INFO] [1743007511.030796397] [rosbag2_recorder]: Pausing recording.
[INFO] [1743007511.030874030] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
[INFO] [1743007511.032019724] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1743007511.032050701] [rosbag2_recorder]: Recording stopped
[INFO] [1743007511.062666560] [rclcpp]: signal_handler(signum=15)

root@tomoyafujita:~/ros2_ws/colcon_ws# ps -ef
UID          PID    PPID  C STIME TTY          TIME CMD
root           1       0  0 09:38 pts/0    00:00:00 /bin/bash
root          19       0  0 09:38 pts/1    00:00:00 /bin/bash
root          93       0  0 09:39 pts/2    00:00:00 /bin/bash
root         160      19  2 09:44 pts/1    00:00:00 /usr/bin/python3 /root/ros2_ws/colcon_ws/install/ros2cli/bin/ros2 topic pub /cmd_vel geometry_msgs/msg/Twist {linear: {x
root         178      93  2 09:44 pts/2    00:00:00 /usr/bin/python3 /root/ros2_ws/colcon_ws/install/ros2cli/bin/ros2 bag record -o test_bag /cmd_vel
root         201       0  0 09:44 pts/3    00:00:00 /bin/bash
root         219     201  0 09:44 pts/3    00:00:00 ps -ef
root@tomoyafujita:~/ros2_ws/colcon_ws# kill -SIGTERM 178
```

> We would like to be able to record the bag and replay it even if the system is hard shutdown

i do not think this can be supported by software 100%.
if the recording data (or any data) is cached on the memory, there is no way to flush the data to the storage when hardware shuts down, unless we have persistent memory used by rosbag2 cache.

for this use case, probably you could use `--max-bag-size` and `--max-cache-size` options to flush the cached data into the storage periodically.

hope this helps, and i will go ahead to remove `bug` tag here.

rowandempster, 2025-04-22:
Hi @EnzoGhisoni @fujitatomoya, I am running into the same issue with our robot power off instantly under normal operation. I am also seeing zero messages available in the resulting .mcap file after booting the robot back up. Is this instant power-off use case not supported by MCAP recording via rosbag2? Or is there a way to instruct rosbag2 to flush every message to disk immediately? 

Thank you,
Rowan

jtbandes, 2025-05-02:
You can also consider reducing your chunk size so that chunks are flushed to disk more frequently which will help give you more recoverable data. Read more here: https://foxglove.dev/blog/understanding-mcap-chunk-size-and-compression

MichaelOrlov, 2025-05-12:
- Closing this issue as WAD (Works As Designed). The problem not on the Rosbag2 side but rather how signals are sent to the container
> you are seeing this problem because you are not sending the signal to the appropriate process ID, but only main process in the container.

closed 2025-05-12 by MichaelOrlov