ros2/rosbag2 #1958: Obey FastDDS XML profile

Ryanf55, 2025-04-03:
## Description

I want to run ROS 2 bag with TCP transport. I already have an XML profile written for FastDDS, but ROS 2 bag uses its own bespoke QoS configuration. This doesn't seem to support TCP transport.
TCP will prevent dropped messages seen in #1845.

## Related Issues

N/A

## Completion Criteria

I can use an XML profile to configure all aspects of FastDDS and ignore the bespoke method that ROS 2 bag has

## Implementation Notes / Suggestions

If we allow subscribers to use SYSTEM_DEFAULT QoS, then an XML profile will apply

## Testing Notes / Suggestions

* Apply an XML profile and observe ability to set reliability QoS on a topic
* Apply an XML profile and observe ability to modify history queue length
* Apply an XML profile and observe ability to use TCP transport
* Run 30Hz image data and 300HZ imu data over TCP transport - observe 0% message loss, and 0% out of order messages

MichaelOrlov, 2025-04-07:
Hi @Ryanf55, You can override QoS settings during recording and replay with Rosbag2, please refer to the relevant section in the readme file https://github.com/ros2/rosbag2/blob/rolling/README.md#overriding-qos-profiles

I am very skeptical that switching to the TCP transport will make a significant difference compared to using a ` reliability: reliable` QoS setting. TCP has a bigger overhead and is known to be heavier and slower.

Anyway, you can give it a try and I would be happy to know your findings.

By the way, we recently added a new feature for replaying from multiple bag files without merging them.  Checked it out, it might be useful in your case.
To play back multiple bags:
```bash
$ ros2 bag play <bag1> -i <bag2> -i <bag3>
```
Messages from all provided bags will be played in order, based on their original recording reception timestamps.

Ryanf55, 2025-04-08:
Hi Michael,

Thanks for the info.

1) ROS 1 uses TCP by default; when we ported to ROS 2, we found that ROS 2 bag with default settings dropped data and recorded out of order samples. Note - for bag recording, we do not necessarily care about bag speed and we can tolerate the overhead because it worked in ROS 1 within our compute constraints.
2) We also found with a custom QoS profile for ROS 2 bag, the same behavior. Not, we could not find how to enable TCP transport on ROS 2 bag.
3) We found that UDP transport with reliable reliability, large history buffers (>100), large kernel buffer, reliable for IMU and image data recording to the same bag works works with <1% drop rate (but not 0). SHM is worse, small kernel buffers are worse, and increased CPU loading from other applications is worse.
4) We also found that we get the best ROS 2 bags if we only run our sensor drivers and ROS 2 bag. Running other CPU intensive applications has a tendency to cause spikes in CPU and dropped data, regardless of our buffer lengths.

We reached out to eProsima for commercial support but have not received a supply. If you are interested to help, please email me at ryan dot friedman at avinc dot com.

With approval from management, I can share some of our plots demonstrating this issue as well as some of our Fast DDS XML profiles and ROS 2 bag QOS  profiles

MichaelOrlov, 2025-04-08:
@Ryanf55 Unfortunately, currently I don't have time to dive deeper in your particular configuration. However, I can try to shed light to better understand what is going on. 
As you already mentioned
> Running other CPU intensive applications has a tendency to cause spikes in CPU and dropped data, regardless of our buffer lengths.

That is true; I observed the same when I was doing experiments when writing my blog post a couple of years ago.
And this could be well explained. In the Rosbag2 we do have to major layers "`transport`" and "`storage`" which is decoupled from each other with double-buffer caching scheme and they are running in a separate threads.  
The double-buffering scheme means that while the transport layer receives messages from the subscriptions and puts them in the one buffer, the storage layer running in a parallel thread taking messages one by one from another buffer and basically finally dumping them to the disk. When the storage buffer becomes empty we lock buffers and switch them by swapping pointers.
It turns out that when CPU load spikes reach close to 100%, the storage thread could go into starvation, which could lead to the overflow of the transport layer buffer. We have no choice in this case; we have to drop messages until the buffer switch happens, and there will be space again in the transport layer buffer.
Another possible scenario is starvation on the Rosbag2 transport layer due to the high CPU load. In this case, messages can be lost on the DDS layer due to internal DDS queue overflows since we are not able to promptly take messages from subscriptions.

Given what I have described and what you have already observed yourself, it makes sense to run the Rosbag2 recorder on dedicated CPU cores isolated from interference from other processes and threads.  
If you don't have a couple of dedicated cores that you can allocate for recording you can consider using a composable recorder node. The main advantage of using composable nodes is "zero-copy" transport. In this case, overall CPU consumption should significantly be reduced. But with the price of running nodes in one process. 
Please be aware that composable recorder became available starting from the Jazzy Jalisco release.

Hope it will help in steering in the right direction.

MichaelOrlov, 2025-04-08:
Also, to reduce extra CPU load, it is recommended to explicitly specify topics for the recorder instead of using regex or the `--all` parameter to avoid running the discovery thread in the background.

Ryanf55, 2025-04-08:
Regarding message drop, what diagnostics are available for each of the two situations? How do we know which one it is on our production builds deployed in the field?

Yep, we will try that. We can experiment with the facilities in the RT kernel.

We're on Jazzy. Composing the recorder for each of the sensor drivers is an option we can try. 

Yep, we specify explicit topics now.

MichaelOrlov, 2025-05-01:
@Ryanf55  Sorry for the late reply.
> Regarding message drop, what diagnostics are available for each of the two situations? How do we know which one it is on our production builds deployed in the field?

As regards a situation when messages are dropped on the DDS side, unfortunately, we don't have any indications about this yet. We do have API in the subscription to be able to specify a callback when messages loss happens. However, nobody has been using it for a while.

As regards the situation messages drop happened on the Rosbag2 side due to the double-buffer overflow we do detect this situation and print warnings when we close the current bag.  
You can look for a message in the log
> Cache buffers lost messages per topic:

Please refer to the corresponding code in the 
https://github.com/ros2/rosbag2/blob/b358c45aa8e25f7d5bdfd1f55970e696ab707b30/rosbag2_cpp/src/rosbag2_cpp/cache/message_cache.cpp#L117-L148

That is, of course, a bit fussy implementation since we don't have a full statistics about lost messages to be available in runtime or saved in the bag file for further analysis. 
That certainly needs to be improved, and this is a candidate for a new feature like "Statistics for lost messages in rosbag".

Perhaps I will be able to allocate time and resources for this feature sometime soon.

MichaelOrlov, 2025-05-12:
- Closing this issue since the question was answered, and the follow-up issue was a feature request was created https://github.com/ros2/rosbag2/issues/2007

closed 2025-05-12 by MichaelOrlov