ros2/rosbag2 #1959: rosbag2_transport Components feature request for Humble

jclinton830, 2025-04-13:
## Description

My understanding is that currently, the ros2_transport package is not support the use of ros2 components for recorder and player. Is there a specific reason for this? 

I am currently working on some project where the hardware supplier does not support Jazzy yet, hence the need for Humble. 

Happy to provide some help with back porting if it is okay.

roncapat, 2025-04-13:
Hi @jclinton830, I was involved in such development. If I recall well, unfortunately backporting would break both API and ABI.

This doesn't mean that it is not possible to achieve it, it's just that it is not supported by the 'humble' branch of this repo.

My starting point, long time ago, before working upstream for implementation of composability, was to create a node component that internally contains a Recorder class. Not sure I can find the original repo though.

roncapat, 2025-04-13:
Here you are: https://github.com/berndpfrommer/rosbag2_composable_recorder

jclinton830, 2025-04-13:
Thanks @roncapat. I will give that a go first. Appreciate it!

MichaelOrlov, 2025-05-01:
- Closing as wouldn't be done, since it requires API/ABI breaking changes.

closed 2025-05-01 by MichaelOrlov