ros2/rosbag2 #1963: ros2 bag play slower than record

biggiantpigeon, 2025-04-17:
I record a ros2 bag about 3.0G, including synchronized lidar/image data at 10Hz and imu data at 200Hz. When I play the bag and watch the timestamps of topics, the time is slower than real world time, and a warning said `[WARN] [1744870148.134381178] [rosbag2_player]: Message queue starved. Messages will be delayed. Consider increasing the --read-ahead-queue-size option.`

 I then realize I have to increase `--read-ahead-queue-size` to 12000 to cover all my 60s data, and add `-p` option to wait it to loading totally, then start -- and everything looks fine. 

I wonder is this a known issue of large data for rosbag2 or I use it wrong? because I convert the bag to ROS1 and it played just fine without any extra options.

here's my version info:
```
root@L-PF4Z9GGC-1415:/home/muge/ros2_ws# dpkg -l | grep ros-humble-rosbag2
ii  ros-humble-rosbag2                                0.15.14-1jammy.20250326.163913          amd64        Meta package for rosbag2 related packages
ii  ros-humble-rosbag2-compression                    0.15.14-1jammy.20250326.160226          amd64        Compression implementations for rosbag2 bags and messages.
ii  ros-humble-rosbag2-compression-zstd               0.15.14-1jammy.20250326.160444          amd64        Zstandard compression library implementation of rosbag2_compression
ii  ros-humble-rosbag2-cpp                            0.15.14-1jammy.20250326.155211          amd64        C++ ROSBag2 client library
ii  ros-humble-rosbag2-interfaces                     0.15.14-1jammy.20250326.153942          amd64        Interface definitions for controlling rosbag2
ii  ros-humble-rosbag2-py                             0.15.14-1jammy.20250326.162756          amd64        Python API for rosbag2
ii  ros-humble-rosbag2-storage                        0.15.14-1jammy.20250326.154257          amd64        ROS2 independent storage format to store serialized ROS2 messages
ii  ros-humble-rosbag2-storage-default-plugins        0.15.14-1jammy.20250326.154858          amd64        ROSBag2 SQLite3 storage plugin
ii  ros-humble-rosbag2-transport                      0.15.14-1jammy.20250326.160658          amd64        Layer encapsulating ROS middleware to allow rosbag2 to be used with or without middleware
```

MichaelOrlov, 2025-04-21:
@biggiantpigeon Unfortunately, when replaying a mix of high-frequency data, such as IMU and slow but heavy camera or Lidar data, there could be a hiccup in data caching due to the suboptimal design in the Player messages queue.
Currently, we have one big queue which is limited by a number of messages, and it reloads in a background thread as soon as getting in size less than specified. However, it could be fast publishing of hundreds of the IMU small messages and then a request to refill the queue with the camera messages. However, refilling with the camera messages may not scale fast enough since need to read big chunks of data from disk.

To mitigate this issue, perhaps the internal player queue representation should be redesigned to have separate queues per topic and limit the queue size by duration rather than by the number of messages.

biggiantpigeon, 2025-05-13:
I just figure out the problem...I'm using windows subsystem(wsl) and this happens if your bag is placed under /mnt/* disk, and will be ok if moved to dir under subsystem. @___@

closed 2025-05-13 by biggiantpigeon