ros2/rosbag2 #1964: `ros2 bag record --action /fibonacci` failed to look up some message types

fujitatomoya, 2025-04-17:
## Description

`ros2 bag record --action /fibonacci` fails to find a few message types to record the action data.

### Expected Behavior

No error detected, all action contents (internal topics and services) can be recorded in the bag file.

### Actual Behavior

Error detected, a few topics are not recorded in the bag file.

## To Reproduce

1. `ros2 run action_tutorials_cpp fibonacci_action_client --ros-args -p action_client_configure_introspection:=contents`
2. `ros2 run action_tutorials_cpp fibonacci_action_server --ros-args -p action_server_configure_introspection:=contents`
3. start recording the action data.

```console
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 bag record --action /fibonacci
[INFO] [1744918458.483621233] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1744918458.487549180] [rosbag2_recorder]: Listening for topics...
[INFO] [1744918458.487558774] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1744918458.487730283] [rosbag2_recorder]: Recording...
[INFO] [1744918458.994384269] [rosbag2_recorder]: Subscribed to topic '/fibonacci/_action/cancel_goal/_service_event'
[ERROR] [1744918458.994473953] [rosbag2_cpp]: Message type name 'example_interfaces/action/Fibonacci_FeedbackMessage' not understood by type definition search, definition will be left empty in bag.
[INFO] [1744918458.996778570] [rosbag2_recorder]: Subscribed to topic '/fibonacci/_action/feedback'
[ERROR] [1744918458.996843436] [rosbag2_cpp]: Message type name 'example_interfaces/action/Fibonacci_GetResult_Event' not understood by type definition search, definition will be left empty in bag.
[INFO] [1744918458.998317877] [rosbag2_recorder]: Subscribed to topic '/fibonacci/_action/get_result/_service_event'
[ERROR] [1744918458.998371474] [rosbag2_cpp]: Message type name 'example_interfaces/action/Fibonacci_SendGoal_Event' not understood by type definition search, definition will be left empty in bag.
[INFO] [1744918458.999690222] [rosbag2_recorder]: Subscribed to topic '/fibonacci/_action/send_goal/_service_event'
[INFO] [1744918459.003753964] [rosbag2_recorder]: Subscribed to topic '/fibonacci/_action/status'
[INFO] [1744918463.673505762] [rosbag2_recorder]: Pausing recording.
[INFO] [1744918463.673559655] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
[INFO] [1744918463.677051924] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1744918463.677080963] [rosbag2_recorder]: Recording stopped
[INFO] [1744918463.746126459] [rclcpp]: signal_handler(signum=2)
```

## Additional context

N.A

fujitatomoya, 2025-04-17:
@Barry-Xu-2018 rosbag2 action finds some errors during recording, can you take a look?

CC: @MichaelOrlov

fujitatomoya, 2025-04-17:
it looks like recording can be completed even with errors.

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 bag info rosbag2_2025_04_17-12_51_37/

Files:             rosbag2_2025_04_17-12_51_37_0.mcap
Bag size:          24.9 KiB
Storage id:        mcap
ROS Distro:        rolling
Duration:          4.014102288s
Start:             Apr 17 2025 12:51:37.944816098 (1744919497.944816098)
End:               Apr 17 2025 12:51:41.958918386 (1744919501.958918386)
Messages:          0
Topic information:
Services:          0
Service information:
Actions:           1
Action information:
  Action: /fibonacci | Type: example_interfaces/action/Fibonacci | Topics: 2 | Service: 3 | Serialization Format: cdr
    Topic: feedback | Count: 3
    Topic: status | Count: 3
    Service: send_goal | Event Count: 6
    Service: cancel_goal | Event Count: 0
    Service: get_result | Event Count: 6
```

christophebedard, 2025-04-17:
I believe actions (`.../action/...` interfaces) are just not supported/expected: https://github.com/ros2/rosbag2/blob/62e5f77b207471c7e3d1b8b5a887910ea079d3d9/rosbag2_cpp/src/rosbag2_cpp/message_definitions/local_message_definition_source.cpp#L52.

From what I understand, this feature is meant for rosbag readers that don't have access to the message definitions (e.g., from an install space), which is usually not the case in ROS 2, so the error is not a big deal.

fujitatomoya, 2025-04-17:
yeah, this is message definition storage integrity. even with error messages, it can record the action data. but something does not add up here, for `.srv` file `encoded_message_definition`...

christophebedard, 2025-04-17:
What do you mean?

fujitatomoya, 2025-04-17:
if `.srv` root type comes in here to get `encoded_message_definition`, it always will be empty definition. it looks like it tries to get the definition recursively, but it never gets. that is what i see now, maybe i am wrong.

christophebedard, 2025-04-17:
These are tests for services, though 🤔 

https://github.com/ros2/rosbag2/blob/62e5f77b207471c7e3d1b8b5a887910ea079d3d9/rosbag2_cpp/test/rosbag2_cpp/test_local_message_definition_source.cpp#L65

fujitatomoya, 2025-04-17:
sorry i was short on the words, this test succeeds. but for `example_interfaces/srv/AddTwoInts_Event` and `example_interfaces/action/Fibonacci_FeedbackMessage`, they never succeed since there are not actual files that it can find. (those types are generated by idl during build time.)

fujitatomoya, 2025-04-17:
I think what needs to be done for now is to support the `/action` with `/srv` here. this brings `/action` to the same level with `/srv` as it is implemented in `LocalMessageDefinitionSource` including recursive `.action` test files. i will patch it right away.

TODO: service / action introspection cases are now excluded to encode type definition encoding in the bag file. this has been already there since service introspection support for rosbag2 with jazzy. since this is not blocking the new feature for `rosbag2 action record/replay` for kilted kaiju, i need to fix this after release including backport to jazzy. for doing this, probably we need to resolve the original `.srv` and `.action` file type definition to be encoded into the storage.

fujitatomoya, 2025-04-17:
@christophebedard @Barry-Xu-2018 @MichaelOrlov what do you think 👆 ?

MichaelOrlov, 2025-04-17:
I agree need to extend `LocalMessageDefinitionSource` to  support finding message definitions for actions

Barry-Xu-2018, 2025-04-18:
LocalMessageDefinitionSource doesn't support action.   

According to https://github.com/ros2/rosbag2/pull/1939#discussion_r2019510232, it still deviates from the original intention according to the current approach. This part needs to be refactored. So I didn't add implementation for action. However, the warning may cause confusion for users.    
Thanks @fujitatomoya, I will check it.

closed 2025-04-18 by fujitatomoya