ros2/rosbag2 #1966: support `.srv` and `.action` generated types for `MessageDefinition::get_full_text`.

fujitatomoya, 2025-04-18:
## Description

https://github.com/ros2/rosbag2/blob/62e5f77b207471c7e3d1b8b5a887910ea079d3d9/rosbag2_cpp/src/rosbag2_cpp/message_definitions/local_message_definition_source.cpp#L208

`LocalMessageDefinitionSource` currently expects that root type file exist in the workspace, for example `ction_msgs/msg/GoalStatusArray`.
This is usually okay if that is general `.msg`, `.srv`, `.action` and `.idl` type files.
But it comes to the problem with generated files from service and action introspection such as `action_msgs/srv/CancelGoal_Event`, `example_interfaces/action/Fibonacci_FeedbackMessage`, and `example_interfaces/action/Fibonacci_GetResult_Event`, because those files do not exist in the workspace.

it eventually ends up raising `DefinitionNotFoundError`, and empty message definition is encoded in the storage.

### Expected Behavior

Originally it tries to encode the message definition information with expanding recursively in the storage, so that rosbag2 can know and manage the bag data with this encoded message definition even if original message files do not exist on the workspace.

That said, it should encode the original `.srv` and `.action` message definition as well.

### Actual Behavior

message definition will be empty for service and action introspection generate types.

## To Reproduce
1. start rosbag2 for action.

```bash
ros2 bag record --action /fibonacci
```

2. `ros2 run action_tutorials_cpp fibonacci_action_client --ros-args -p action_client_configure_introspection:=contents`
3. `ros2 run action_tutorials_py fibonacci_action_server --ros-args -p action_server_configure_introspection:=contents`

## Additional context

see https://github.com/ros2/rosbag2/issues/1964#issuecomment-2814187537

- we can break down the 2 PRs. the one is service, the other is action. so that we can backport service PR to jazzy.
- test also needs to be added with service and action introspection names to be resolved to original service and action message.

fujitatomoya, 2025-04-18:
@Barry-Xu-2018 i just created the follow-up issue here. please add more information or thoughts if you have.

Barry-Xu-2018, 2025-04-18:
I will handle these two PRs. Please assign this issue to me.

Barry-Xu-2018, 2025-04-23:
@fujitatomoya @MichaelOrlov 

I would like to hear your opinions.

The current interface, LocalMessageDefinitionSource::get_full_text, only takes one parameter
```c++
rosbag2_storage::MessageDefinition LocalMessageDefinitionSource::get_full_text(
  const std::string & root_type)
```
For the topics related to action
```
/fibonacci/_action/cancel_goal/_service_event [action_msgs/srv/CancelGoal_Event]
/fibonacci/_action/feedback [example_interfaces/action/Fibonacci_FeedbackMessage]
/fibonacci/_action/get_result/_service_event [example_interfaces/action/Fibonacci_GetResult_Event]
/fibonacci/_action/send_goal/_service_event [example_interfaces/action/Fibonacci_SendGoal_Event]
/fibonacci/_action/status [action_msgs/msg/GoalStatusArray]
```
It is not possible to extract the action type from the topic types `cancel_goal` and `status`.   
If we only modify get_full_text() (without changing the upper-level code), it is not possible for all five related topics of action to return the action type information.  

To ensure that the type passed down is the actual type (service type/action type), I plan to obtain the real type within RecorderImpl::subscribe_topics(). To store the actual type, I need to extend the TopicMetadata struct by adding a field to hold the actual type. When calling get_full_text() in SequentialWriter::create_topic(), the actual type will be passed directly.

Implementation codes like https://github.com/ros2/rosbag2/compare/rolling...Barry-Xu-2018:rosbag2:develop/topic-fix-action-message-definition-issue

Since these changes require modifying the interface, I would like to hear your opinions.  

This approach cannot be used for the Jazzy version to fix the service issue, since the interface has been modified. Therefore, I have created a separate patch (https://github.com/ros2/rosbag2/pull/1968).

Barry-Xu-2018, 2025-04-28:
Friendly ping @fujitatomoya @MichaelOrlov  

Please take a look at my above comment https://github.com/ros2/rosbag2/issues/1966#issuecomment-2823205511.   
I'd like to hear your opinions.

fujitatomoya, 2025-04-29:
@Barry-Xu-2018 sorry for being late. i will look into the details for tomorrow.

off the top of my head (my assumption),

this encoded type text information is meant to be used to decode the messages on the system where does not have the message type. if that is true, original `msg`, `srv` and `action` file needs to be encoded as text in this field? so that this information can be decoded later?
but i am not sure how it takes advantage of this information as user perspective. for example, it can create the original message file based on this encoded information and generate the type support then use it when playing the bag file? if that is so, maybe we can store the original `srv` and `action` file for this field. but it is used in other way, i am not sure saving the original files in this field is the right thing to do.

Barry-Xu-2018, 2025-04-29:
@fujitatomoya 

> off the top of my head (my assumption),
>
> this encoded type text information is meant to be used to decode the messages on the system where does not have the message type. if that is true, original [...]

The current implementation is to store the original srv and action files. My understanding is that the original message type support can be reconstructed based on the saved message definition files.

MichaelOrlov, 2025-04-29:
@Barry-Xu-2018 @fujitatomoya Please refer to the unfinished [Dynamic Subscription (REP-2011 subset) #1374](https://github.com/ros2/ros2/issues/1374), which is supposed to add support for creating subscriptions from message definitions stored in the bag files without generating type supports at compile time.

MichaelOrlov, 2025-04-29:
@Barry-Xu-2018 The main point in saving message definition is that we shall be able to introspect a serialized message in the future using the stored message definition on a system which doesn't have compile time type supports for those message definitions or when message definition evolved over a time.

The message introspection in particularly well described in the 
 [ROS 2 REP-2011 in section Run-Time Interface Reflection](https://github.com/ros-infrastructure/rep/pull/358/files#diff-af97236a4092c2373a4209a351966da2a6b4f873301bf907d46358680405be49R339-R407)

Please refer also to the [REP-2016 - Type Description Distribution (REP-2011 subset / REP-2016) - formerly "message definitions on-the-wire"](https://github.com/ros2/ros2/issues/1159) for more details.

Barry-Xu-2018, 2025-04-30:
@MichaelOrlov 

Thank you for your guidance.  
According to saving message definition (which can generate `rosidl_runtime_c__type_description__TypeDescription`), dynamic typesupport can be created. And then DynamicSubscription can use this dynamic typesupport.  DynamicService/DynamicAction has not been implemented yet.  
In order to support Dynamic typesupport (Even though DynamicSubscription/DynamicService/DynamicAction hasn't been implemented yet.), we should save message, service and action definition in rosbag.   

If I have misunderstood, please correct me.

MichaelOrlov, 2025-04-30:
@Barry-Xu-2018 
> we should save message, service and action definition in rosbag.

I think that is correct.

fujitatomoya, 2025-05-02:
@Barry-Xu-2018 

> In order to support Dynamic typesupport (Even though DynamicSubscription/DynamicService/DynamicAction hasn't been implemented yet.), we should save message, service and action definition in rosbag.

i am still not sure how exactly this is going to be implemented. but what we have to do here for the future development of `DynamicService` and `DynamicAction` is that saving the original `.srv` and `.action` files data in the encoded field, right?

> It is not possible to extract the action type from the topic types cancel_goal and status.

and coming back to the original question here. how we can look up the original `.action` type (`.srv` is easy) from those topics. i think we eventually need to some association from the layer that manages the actions? sorry, i do not have good proposal here.

fujitatomoya, 2025-05-06:
> Yes. I made modifications (https://github.com/ros2/rosbag2/pull/1968) at a higher level to ensure that the information for all action topics is correctly saved. But this will break the interface.

i mean, we can do that, right? at least for rolling unless there is a way around.

Barry-Xu-2018, 2025-05-07:
> i mean, we can do that, right? at least for rolling unless there is a way around.

Yes. 

@fujitatomoya @MichaelOrlov 

How can this issue be addressed currently?

If the interface isn't changed
- https://github.com/ros2/rosbag2/pull/1968 can be applied to `Jazzy`, `Kalited` to resolve service message definition issue.
- About action message definition issue, we only make sure `send_goal/get_result/status` get correct action message. But `cancel_goal and status` cannot get action message.  So for `Kalited`, how to do?  1. keep current situation (No message definition for action interface topic). 2. Only return action message for `send_goal/get_result/status` and return nothing for `cancel_goal and status` 

For `rolling`, we can change interface to completely resolve the issue with action (Like https://github.com/ros2/rosbag2/pull/1968). 

I'd like to hear your thoughts.

Barry-Xu-2018, 2025-05-13:
Friendly ping @MichaelOrlov.   
Please take a look at this comment https://github.com/ros2/rosbag2/issues/1966#issuecomment-2857692563 and share you thought. I want to know how to continue.

fujitatomoya, 2025-05-14:
@Barry-Xu-2018 this is on my list to do, i will take a look at the draft PR maybe next week.

MichaelOrlov, 2025-06-04:
@Barry-Xu-2018 I lost context in your original question.  
Could you please clarify why we can't get the message definition for the `cancel_goal` and `status` in the action?

Also, as regards the 
> For rolling, we can change interface to completely resolve the issue with action (Like https://github.com/ros2/rosbag2/pull/1968).

Can you please clarify what API needs to be changed, and how are you going to solve it for Rolling?  
It is unclear from the current state of the https://github.com/ros2/rosbag2/pull/1968

MichaelOrlov, 2025-06-05:
@fujitatomoya @Barry-Xu-2018 
As regards the original question of how to avoid API/ABI breaking changes and proceed with implementation for Kilted.
I have spent some time, and it seems I have found the solution. 
- Please see my draft PR https://github.com/Barry-Xu-2018/rosbag2/pull/8

Barry-Xu-2018, 2025-06-05:
@MichaelOrlov 

> Could you please clarify why we can't get the message definition for the cancel_goal and status in the action？

This function is responsible for obtaining the message definition, but the interface only provides the topic type.
https://github.com/ros2/rosbag2/blob/62e5f77b207471c7e3d1b8b5a887910ea079d3d9/rosbag2_cpp/src/rosbag2_cpp/message_definitions/local_message_definition_source.cpp#L208-L209

For `cancel_goal` and `status`, no matter what the action is, the messages for these two action interfaces are fixed (action_msgs/srv/CancelGoal_Event and action_msgs/msg/GoalStatusArray). So without additional information, it's impossible to know which action they belong to.

>> For rolling, we can change interface to completely resolve the issue with action (Like https://github.com/ros2/rosbag2/pull/1968).
>
> Can you please clarify what API needs to be changed, and how are you going to solve it for Rolling?
It is unclear from the current state of the https://github.com/ros2/rosbag2/pull/1968

#1968 only resolved the issue with the service's message definition.  
To address the problem with the action's message definition, it's still necessary to modify the parameters of `LocalMessageDefinitionSource::get_full_text()` to provide more information.

> As regards the original question of how to avoid API/ABI breaking changes and proceed with implementation for Kilted.
I have spent some time, and it seems I have found the solution.
>
> Please see my draft PR https://github.com/Barry-Xu-2018/rosbag2/pull/8

Thanks for the suggestion. By this way, we can avoid modifying the existing interface.   
I will continue working on this PR.

Barry-Xu-2018, 2025-06-06:
@MichaelOrlov 

> That is the main difference, your version with interface modification is not backportable to the Kilted due to the API/ABi breaking changes. Mine doesn't have API/ABi breaking changes and could be backported to the Kilted distro.

Yes.   
But I found an issue for https://github.com/Barry-Xu-2018/rosbag2/tree/review/topic-fix-action-message-definition-issue with your modification.
For goal_cancel service topic and status topic, we still cannot obtain the correct action message type.  
Even with the information on topic name and topic type, we still cannot convert it to an action message type.  
for example
```
/fibonacci/_action/cancel_goal/_service_event [action_msgs/srv/CancelGoal_Event]
/fibonacci/_action/status [action_msgs/msg/GoalStatusArray]
```
According to above information, we cannot get action type `example_interfaces/action/Fibonacci`.  
Therefore, simply modifying `MessageDefinition get_full_text()` cannot solve the problem.

To resolve it, changes need to be made to the higher-level code.  

Here's another way: instead of changing `MessageDefinition get_full_text()`, I obtain the actual type in RecorderImpl::subscribe_topics().   
https://github.com/Barry-Xu-2018/rosbag2/commit/425620474c6252846830df6d1ab9fc6d4cdfbe5b
The downside is that this modification will break API/ABI compatibility and cannot be backported to Kilted.

Barry-Xu-2018, 2025-06-13:
@MichaelOrlov 

> I am still against doing a workaround in a hacky way on the upper rosbag2_transport layer and adding a new field to the TopicMetadata struct just to serve one particular case with action introspections.

I understand. That's just my thought. 

> - I created a https://github.com/Barry-Xu-2018/rosbag2/pull/9 PR as an example of how I would envision the same changes, but made in a clean and concise way without changes on the upper rosbag2_transport layer.
> - The ABI changes are unavoidable since we need to add a map for caching the inner action types to the publicly exposed header file. However, this fix will not have API-breaking changes.
> - My assumption is that we can backport changes from the https://github.com/Barry-Xu-2018/rosbag2/tree/review/topic-fix-action-message-definition-issue, which relates to replacing get_full_text(topic_type) with get_full_text_ext(topic_name, topic_type) to the Kilted. However, the other https://github.com/Barry-Xu-2018/rosbag2/pull/9 PR suggested by me can go only to the Rolling since it has an ABI-breaking changes.

I totally agree with your idea.  
At first, I will create a PR based on https://github.com/Barry-Xu-2018/rosbag2/tree/review/topic-fix-action-message-definition-issue for rolling (There are still some test codes that need updating, and I will make the necessary changes.). This PR can be backported to Kilted.   
And then create a new PR based on https://github.com/Barry-Xu-2018/rosbag2/pull/9 for rolling.

Barry-Xu-2018, 2025-06-16:
@MichaelOrlov 

> At first, I will create a PR based on https://github.com/Barry-Xu-2018/rosbag2/tree/review/topic-fix-action-message-definition-issue for rolling (There are still some test codes that need updating, and I will make the necessary changes.). This PR can be backported to Kilted.

I have created a new PR #2041.

MichaelOrlov, 2025-06-25:
> [@MichaelOrlov](https://github.com/MichaelOrlov) [@Barry-Xu-2018](https://github.com/Barry-Xu-2018) thanks for closing this issue!

@fujitatomoya I promise you, there will be a few more follow-ups. 😉

PR #2041 (2025-06-23): Fix service/action message definition issue
## Description

Fixes #1966  

This fix can be backported to Kilted. (API/ABI Compatible)

### Is this user-facing behavior change?

### Did you use Generative AI?

No

### Additional Information

None