ros2/rosbag2 #1973: ros2 bag convert data loss

DangitBen, 2025-04-22:
## Description
`ros2 bag convert` has data loss when converting to a bag using `compression_mode: message`

### Expected Behavior

- The compressed and uncompressed versions of bags should contain the same messages.
- `ros2 bag info uncompressed`, `ros2 bag info compressed-message`, and `ros2 bag info compressed-file` should output the same message counts for each topic 

### Actual Behavior

```bash
$ ros2 bag info uncompressed

Files:             uncompressed_0.mcap
Bag size:          4.9 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          0.000000029s
Start:             Dec 31 1969 18:00:00.000000000 (0.000000000)
End:               Dec 31 1969 18:00:00.000000029 (0.000000029)
Messages:          30
Topic information: Topic: /strings | Type: std_msgs/msg/String | Count: 30 | Serialization Format: cdr
Service:           0
Service information:
```

Example output I received: 
```bash
$ ros2 bag info compressed-file

Files:             compressed-file_0.mcap.zstd
Bag size:          1.8 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          0.000000029s
Start:             Dec 31 1969 18:00:00.000000000 (0.000000000)
End:               Dec 31 1969 18:00:00.000000029 (0.000000029)
Messages:          30 # looks good!
Topic information: Topic: /strings | Type: std_msgs/msg/String | Count: 30 | Serialization Format: cdr
Service:           0
Service information:
```
So far so good for file compression mode. But message compression mode has a problem: 

```bash
$ ros2 bag info compressed-message

Files:             compressed-message_0.mcap
Bag size:          5.1 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          0.000000029s
Start:             Dec 31 1969 18:00:00.000000000 (0.000000000)
End:               Dec 31 1969 18:00:00.000000029 (0.000000029)
Messages:          28 # oh no
Topic information: Topic: /strings | Type: std_msgs/msg/String | Count: 28 | Serialization Format: cdr
Service:           0
Service information:
```

Number of messages dropped is inconsistent between attempts. Here is the output after attempting the conversion again clean:
```bash
$ ros2 bag info compressed-message

Files:             compressed-message_0.mcap
Bag size:          5.2 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          0.000000029s
Start:             Dec 31 1969 18:00:00.000000000 (0.000000000)
End:               Dec 31 1969 18:00:00.000000029 (0.000000029)
Messages:          29 # inconsistent!
Topic information: Topic: /strings | Type: std_msgs/msg/String | Count: 29 | Serialization Format: cdr
Service:           0
Service information:
```

## To Reproduce

1. Collect a bag or generate.
    - Script attached below to generate uncompressed bag used above.
    - Issue has appeared for both real/simulated data.
2. Create yaml used for compression `compressed.yaml`:
```yaml
output_bags:
- uri: compressed-file
  storage_id: mcap
  all_topics: true
  all_services: true
  compression_mode: file
  compression_format: zstd
- uri: compressed-message
  storage_id: mcap
  all_topics: true
  all_services: true
  compression_mode: message
  compression_format: zstd

```
3. `ros2 bag convert -i uncompressed -o compressed.yaml`
4. `ros2 bag info uncompressed`
5. `ros2 bag info compressed-file`
5. `ros2 bag info compressed-message`

## System
 - OS: Ubuntu Noble
 - ROS 2 Distro: Jazzy
 - Install Method: APT
 - Version: `0.26.6-1noble.20250406.110400`

## Additional context

I tested on 22.04/Humble using `sqlite3` and saw the same issue.

Script to create uncompressed bag used above.
```python
import rosbag2_py
from rclpy.serialization import serialize_message
from std_msgs.msg import String
writer = rosbag2_py.SequentialWriter()
writer.open(
    rosbag2_py.StorageOptions(uri="uncompressed", storage_id="mcap"),
    # rosbag2_py.StorageOptions(uri="uncompressed", storage_id="sqlite3"), # storage doesn't matter either
    rosbag2_py.ConverterOptions(),
)

writer.create_topic(rosbag2_py.TopicMetadata(0, "/strings", "std_msgs/msg/String", "cdr"))

for i in range(30):
    msg_out = String()
    msg_out.data = "Test1"
    writer.write("/strings", serialize_message(msg_out), i)
    # writer.write("/strings", serialize_message(msg_out), 0) # timestamp doesn't seem to matter for this
writer.close()
```

MichaelOrlov, 2025-04-22:
@DangitBen Can you please try to explicitly setup `compression_queue_size = 0` in your `compressed.yaml` file to see if it will solve the problem?

MichaelOrlov, 2025-04-22:
Usually, we are dropping messages during compression here
https://github.com/ros2/rosbag2/blob/c712dc24ca54bd65f9836cc3ebd8c11036e94350/rosbag2_compression/src/rosbag2_compression/sequential_compression_writer.cpp#L422-L437

in `compressor_message_queue_.pop(); `

DangitBen, 2025-04-22:
@MichaelOrlov Yes, that seems to do it!

```yaml
output_bags:
- uri: compressed-file
  storage_id: mcap
  all_topics: true
  all_services: true
  compression_mode: file
  compression_format: zstd
  compression_queue_size: 0
- uri: compressed-message
  storage_id: mcap
  all_topics: true
  all_services: true
  compression_mode: message
  compression_format: zstd
  compression_queue_size: 0
```

DangitBen, 2025-04-22:
I see that there are related/duplicate issues that I was unable to find when trying to debug this myself. (searching for compression_queue_size brought them up)

#1119 and #1058 especially.

Is there a reason  `compression_queue_size` isn't set to 0 by default for `ros2 bag convert`? It seems strange to have a default that can result in silently lost data.

MichaelOrlov, 2025-04-23:
> Is there a reason compression_queue_size isn't set to 0 by default for ros2 bag convert? It seems strange to have a default that can result in silently lost data.

That is a good question. I was pretty much sure that we initialized it to **0** in the `ros2 bag convert`, need to double check and investigate if it is not.

DangitBen, 2025-04-24:
> That is a good question. I was pretty much sure that we initialized it to **0** in the `ros2 bag convert`, need to double check and investigate if it is not.

I think I've figured out what's happening. Yes, there is a default set here on 503 (via the `bag_rewrite_default_record_options` function) but it is overwritten on 504:
https://github.com/ros2/rosbag2/blob/dcaca47599611a378d20fc897fdfc14df8813c92/rosbag2_py/src/rosbag2_py/_transport.cpp#L503-L504

`YAML::Node.as<rosbag2_transport::RecordOptions>` calls `YAML::convert<rosbag2_transport::RecordOptions>` with a default constructed `RecordOptions`, not the one with modified defaults on 503.

I'd recommend replacing 504 with something similar to what is done on 502 (though I haven't tested it):
```cpp
YAML::convert<rosbag2_storage::RecordOptions>::decode(bag_node, record_options);
```

DangitBen, 2025-04-24:
In fact, it was originally implemented that way on 5b8b658, but was changed for some reason on abdc408

MichaelOrlov, 2025-05-01:
- Closing as fixed in the #1985

closed 2025-05-01 by MichaelOrlov