ros2/rosbag2 #1994: Inconsistent Snapshot Service Behavior Across Systems

manan-peppermint, 2025-04-28:
While using the /rosbag2_recorder/snapshot service, I observed different behaviors on two different systems:

    On System A: Every time I call the snapshot service, a new MCAP file is created inside the output directory (if the folder already exists).

    On System B: Every time I call the snapshot service, the existing MCAP file is overwritten if the output directory already exists.

This inconsistency makes it difficult to design a predictable snapshot-saving workflow across systems. Is there any flag to decide the behavior of output when doing service call.

MichaelOrlov, 2025-05-01:
@manan-peppermint, Can you please provide more information on how to reproduce this issue?
What ROS Distro to use and version?
What OS?
Exact steps to reproduce?

manan-peppermint, 2025-05-02:
I am using ROS 2 Humble on Ubuntu 22.04.

I'm making a client call to the /rosbag2_recorder/snapshot service in a C++ node when a certain condition is met. The client call code looks like this:
```
auto request = std::make_shared<rosbag2_interfaces::srv::Snapshot::Request>();
auto future_result = rosbag_snapshot_client->async_send_request(
    request,
    std::bind(&RecordSnapshot::responseCb, this, std::placeholders::_1)
);
```
I'm starting the snapshot recording service using subprocess module for python.

I have deployed the same set of packages on both systems, but I’m observing different behavior between them.

Can you please help identify what could cause this discrepancy?
Let me know if you need additional logs or steps to reproduce the issue.

manan-peppermint, 2025-05-02:
On system A the output get overwritten in the same file again and again:

[image]

On system B the output is stored in a new file:

[image]

Both this system have the same ros2 package.

MichaelOrlov, 2025-05-05:
@emersonknapp Since you are also using Humble, could you please help triage this issue?