ros2/rosbag2 #2002: ROS2 services not in bagfile

Ryanf55, 2025-05-07:
## Description

Using ROS 2 bag record, when attempting to record services, they don't show up in the bag.

### Expected Behavior

With the `--services` flag when recording, I expect my services to show up.

### Actual Behavior

The bag file does not contain services.

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Run in the first terminal: `ros2 run demo_nodes_py add_two_ints_server`
2. Run in a second terminal: `ros2 bag record --services /add_two_ints`
3. Run in a third terminal: `ros2 run demo_nodes_py add_two_ints_client`
4. Observe the client add two ints and get a result of `5`.
5. In the second terminal, hit ctrl+C, and observe the logs:
```
[INFO] [1746630690.545016248] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1746630690.596575255] [rosbag2_recorder]: Listening for topics...
[INFO] [1746630690.596609442] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1746630690.596876563] [rosbag2_recorder]: Recording...
[INFO] [1746630725.332908712] [rosbag2_recorder]: Pausing recording.
[INFO] [1746630725.333013146] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
[INFO] [1746630725.333895371] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1746630725.333959242] [rosbag2_recorder]: Recording stopped
[INFO] [1746630725.596161967] [rclcpp]: signal_handler(signum=2)
```
6. In the second terminal, inspect the bag.
```
$ ros2 bag info rosbag2_2025_05_07-15_11_30/

Files:             rosbag2_2025_05_07-15_11_30_0.mcap
Bag size:          1.9 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          0.000000000s
Start:             Apr 11 2262 23:47:16.854775807 (9223372036.854775807)
End:               Apr 11 2262 23:47:16.854775807 (9223372036.854775807)
Messages:          0
Topic information: 
Service:           0
Service information: 
```

## System (please complete the following information)
 - OS: Ubuntu Noble (24)
 - ROS 2 Distro: Jazzy
 - Install Method: apt
 - Version: 0.26.7-1noble.20250430.024741
 - Middleware: Default FastDDS

## Additional context
** Add any other context about the problem here **

fujitatomoya, 2025-05-07:
@Ryanf55 to record or `ros2 service echo` the service data, we need to enable service introspection on the service client and server. please have a look at https://docs.ros.org/en/jazzy/Tutorials/Demos/Service-Introspection.html

Ryanf55, 2025-05-07:
Nice, RTFM :)

closed 2025-05-07 by Ryanf55