ros2/rosbag2 #2003: ros2 bag reindex does not recreate message index

peci1, 2025-05-08:
## Description
`ros2 bag reindex` does not recreate message index in bags whose recording was terminated forcefully.

### Expected Behavior
When reindex successfully finishes, I'd expect the bag to be 100% usable, same as bags recorded fully.

### Actual Behavior
```
$ ros2 bag reindex rosbag2_2025_05_08-14_09_36                                                                                                                                                              
[INFO] [1746706244.158341159] [rosbag2_cpp]: Beginning reindexing bag in directory: "rosbag2_2025_05_08-14_09_36"       
[WARN] [1746706244.207632544] [rosbag2_storage_mcap]: attempted to read in receive timestamp order with no message index                                                                                    
[WARN] [1746706244.207654755] [rosbag2_cpp]: Could not set read order on open(), falling back to file order                                                                                                 
[INFO] [1746706244.216940004] [rosbag2_cpp]: Reindexing complete.                                                                                                                                           
$ ros2 run rqt_bag rqt_bag rosbag2_2025_05_08-14_09_36                                                                                                                                                      
Qt: Session management error: None of the authentication protocols specified are supported                                                                                                                  
[WARN] [1746706250.787082517] [rosbag2_storage_mcap]: attempted to read in receive timestamp order with no message index                                                                                    
[WARN] [1746706250.862940743] [rosbag2_storage_mcap]: attempted to read in receive timestamp order with no message index                                                                                    
[WARN] [1746706250.866858055] [rosbag2_storage_mcap]: attempted to read in receive timestamp order with no message index                                                                                    
[WARN] [1746706250.870472811] [rosbag2_storage_mcap]: attempted to read in receive timestamp order with no message index
[WARN] [1746706250.875138264] [rosbag2_storage_mcap]: attempted to read in receive timestamp order with no message index
[WARN] [1746706250.881880333] [rosbag2_storage_mcap]: attempted to read in receive timestamp order with no message index      
```

## To Reproduce
1. Play a well working bag
2. `ros2 bag record -a` in another console
3. `kill -9` the recording node
4. `ros2 bag reindex $bag`
5. `ros2 run rqt_bag rqt_bag $bag`

## System (please complete the following information)
 - OS: Ubuntu Noble
 - ROS 2 Distro: Jazzy
 - Install Method: APT
 - Version: release

## Additional context
Bag recording can be forcefully terminated for various reasons, e.g. voltage spikes on power distribution which reboot the PC, batteries running out etc.

MichaelOrlov, 2025-07-11:
- Closing this issue as WAD (Works As Defined)

peci1, 2025-07-17:
@MichaelOrlov Are you sure with your decision about WAD? If so, I'll have to add this to my ROS2 pain points list. Because:

1. It works different from how ROS 1 worked (it reindexed the storage).
2. It tells that the index has been recreated, but other tools tell the user that the index is missing. It's not easy to learn there are actually multiple types of indices.

> because it is in pugins and upper layer don't know internal spicififcs about how to recover the broken file.

Then there should be a plugin API for reindexing the storage. I think it is not a big extrapolation to say that each storage has some kind of index. So it would make perfect sense if the storage plugin could be told to rebuild the index (whatever it is). And, of course, then it would also make sense to have an API to check the index validity.

peci1, 2025-07-17:
> `mcap recover bag_file`

This works, but you need to save the storage to a different file.