ros2/rosbag2 #2004: `ros2 bag play --send-actions-as-client` includes non-exists cancel request

fujitatomoya, 2025-05-09:
## Description

after recording action data with `ros2 bag record --action /fibonacci`, bag information does not include any cancel service communication at all. this bag information looks no problem at all. but once we play the action data from this bag file with `ros2 bag play --send-actions-as-client`, we can see cancel request with `ros2 action echo`. that is strange because bag file does not have any cancel service data for the action.

### Expected Behavior

cancel request and response should not occur during playback.

### Actual Behavior

cancen request and response does exist when bag file is played.

## To Reproduce

1. `ros2 bag record --action /fibonacci`
2. `ros2 run action_tutorials_cpp fibonacci_action_server --ros-args -p action_server_configure_introspection:=contents`
3. `ros2 run action_tutorials_py fibonacci_action_client --ros-args -p action_client_configure_introspection:=contents`
4. `ros2 bag info <bag file>` to make sure cancel request does not exist in the bag file.

```console
root@tomoyafujita:~/docker_ws/ros2_colcon# ros2 bag info rosbag2_2025_05_08-20_21_37/

Files:             rosbag2_2025_05_08-20_21_37_0.mcap
Bag size:          20.6 KiB
Storage id:        mcap
ROS Distro:        rolling
Duration:          9.002494437s
Start:             May  8 2025 20:21:42.206390139 (1746760902.206390139)
End:               May  8 2025 20:21:51.208884576 (1746760911.208884576)
Messages:          0
Topic information: 
Services:          0
Service information: 
Actions:           1
Action information: 
  Action: /fibonacci | Type: example_interfaces/action/Fibonacci | Topics: 2 | Service: 3 | Serialization Format: cdr
    Topic: feedback | Count: 9
    Topic: status | Count: 2
    Service: send_goal | Event Count: 4
    Service: cancel_goal | Event Count: 0
    Service: get_result | Event Count: 4
```

5. `ros2 action echo /fibonacci example_interfaces/action/Fibonacci --flow-style`
6. `ros2 bag play <bag file>`
7. You can now see in the terminal where `ros2 action echo` is running. And that should not exist.

```console
...
---
interface: CANCEL_SERVICE
info:
  event_type: REQUEST_RECEIVED
  stamp:
    sec: 1746763444
    nanosec: 628895364
  client_gid: [1, 15, 227, 174, 73, 171, 224, 133, 0, 0, 0, 0, 0, 0, 21, 4]
  sequence_number: 1
request: [{goal_info: {goal_id: {uuid: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]}, stamp: {sec: 0, nanosec: 0}}}]
response: []
---
interface: CANCEL_SERVICE
info:
  event_type: RESPONSE_SENT
  stamp:
    sec: 1746763444
    nanosec: 629197756
  client_gid: [1, 15, 227, 174, 73, 171, 224, 133, 0, 0, 0, 0, 0, 0, 21, 4]
  sequence_number: 1
request: []
response: [{return_code: 0, goals_canceling: [{goal_id: {uuid: [49, 202, 53, 245, 220, 95, 202, 221, 142, 251, 241, 255, 106, 118, 102, 171]}, stamp: {sec: 1746763435, nanosec: 702373689}}]}]
---
...
```

## Additional context

N.A

fujitatomoya, 2025-05-09:
@Barry-Xu-2018 any thoughts?

MichaelOrlov, 2025-05-12:
- Closing as fixed in the https://github.com/ros2/rosbag2/pull/2005

closed 2025-05-12 by MichaelOrlov