ros2/rosbag2 #2013: Humble: ability to use start/end time with bag convert

andanders, 2025-05-14:
## Description

`ros2 bag convert` has the ability to split the bag by start and end time. Just not in Humble.
And if you use Jazzy to do the conversion that changes the bag so Humble cannot use it.

Example output.yaml
```yaml
output_bags:
- uri: converted
  storage_id: "sqlite3"
  all_topics: true
  start_time_ns: 1747050866000000000
  end_time_ns:   1747050928000000000
```

After converting bag with Jazzy and running `ros2 bag info` or `play` in Humble:
`Exception on parsing info file: yaml-cpp: error at line 15, column 11: bad conversion`

MichaelOrlov, 2025-05-24:
@andanders We can't backport changes which will introduce API/ABI breaking changes - this is why feature start/end time in bag convert are not available on Humble.

You see error `Exception on parsing info file: yaml-cpp: error at line 15, column 11: bad conversion` after converting the bag with Jazzy and running ros2 bag info or play in Humble, because we changed the format of how we serialize the `offered_qos_profiles`.
While we can't backport https://github.com/ros2/rosbag2/pull/1476 as is, it could be possible to partially backport it to support a new QoS format on Humble.
If someone could take this task, it would be great.
Also, you can find more details in the https://github.com/ros2/rosbag2/issues/1895#issuecomment-2722758287

MichaelOrlov, 2026-02-05:
Closing as wouldn't be done since we can't add a new fields 
```
  start_time_ns:
  end_time_ns:
```
to the `StorageOptions` struct on `Humble` because it will break API/ABI compatibility.