ros2/rosbag2 #2014: 🧑‍🌾 `RecordIntegrationTestFixture` - `test_record_all` flaky failure

Crola1702, 2025-05-19:
## Description
`test_record_all` test is having flaky behavior in nightly CI

Test regressions:
* [rosbag2_transport.RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded](https://ci.ros2.org/view/nightly/job/nightly_linux_release/3479/testReport/junit/rosbag2_transport/RecordIntegrationTestFixture/published_messages_from_multiple_topics_are_recorded/)
* [rosbag2_transport.RecordIntegrationTestFixture.published_messages_from_topic_service_action_are_recorded](https://ci.ros2.org/view/nightly/job/nightly_linux_release/3479/testReport/junit/rosbag2_transport/RecordIntegrationTestFixture/published_messages_from_topic_service_action_are_recorded/)

### Expected Behavior
Test should pass

### Actual Behavior
Test should fail

## System (please complete the following information)
 - OS: All CI distros
 - ROS 2 Distro: Rolling
 - Install Method: Source build
 - Version: rolling

## Additional context

Reference builds:
- https://ci.ros2.org/view/nightly/job/nightly_linux_release/3479/
- https://ci.ros2.org/job/nightly_linux-aarch64_release/3109/
- https://ci.ros2.org/job/nightly_win_rep/3739/
- https://ci.ros2.org/job/nightly_linux_coverage/2779
- https://ci.ros2.org/job/nightly_linux-rhel_release/2275/

Test regressions:
* [rosbag2_transport.RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded](https://ci.ros2.org/view/nightly/job/nightly_linux_release/3479/testReport/junit/rosbag2_transport/RecordIntegrationTestFixture/published_messages_from_multiple_topics_are_recorded/)
* [rosbag2_transport.RecordIntegrationTestFixture.published_messages_from_topic_service_action_are_recorded](https://ci.ros2.org/view/nightly/job/nightly_linux_release/3479/testReport/junit/rosbag2_transport/RecordIntegrationTestFixture/published_messages_from_topic_service_action_are_recorded/)

Log output:
```
/home/jenkins-agent/workspace/nightly_linux_release/ws/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_record_all.cpp:81
Expected equality of these values:
  recorded_messages.size()
    Which is: 5
  expected_messages
    Which is: 4
...
/home/jenkins-agent/workspace/nightly_linux_release/ws/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_record_all.cpp:262
Expected equality of these values:
  recorded_messages.size()
    Which is: 16
  expected_messages
    Which is: 11
```

<details>
<summary>Flaky ratio last 15 days:</summary>

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_win_rep|2025-05-19|2025-05-19|10|1|10.0
nightly_linux_release|2025-05-19|2025-05-12|15|4|26.67
nightly_linux_coverage|2025-05-19|2025-05-19|11|1|9.09
nightly_linux-rhel_release|2025-05-19|2025-05-06|15|2|13.33
nightly_linux-rhel_debug|2025-05-19|2025-05-05|14|4|28.57
nightly_linux-aarch64_repeated|2025-05-19|2025-05-06|13|4|30.77
nightly_linux-aarch64_release|2025-05-19|2025-05-06|15|5|33.33
Rci__nightly-connext_ubuntu_noble_amd64|2025-05-19|2025-05-06|10|5|50.0
Rci__nightly-release_ubuntu_noble_amd64|2025-05-18|2025-05-15|5|2|40.0
Rci__nightly-debug_ubuntu_noble_amd64|2025-05-16|2025-05-16|4|1|25.0
Kci__nightly-connext_ubuntu_noble_amd64|2025-05-16|2025-05-16|2|1|50.0
nightly_linux-aarch64_debug|2025-05-15|2025-05-15|15|1|6.67
nightly_linux_repeated|2025-05-14|2025-05-14|12|1|8.33
nightly_linux-rhel_repeated|2025-05-14|2025-05-14|13|1|7.69
nightly_linux_debug|2025-05-13|2025-05-05|15|4|26.67

</details>

MichaelOrlov, 2025-05-22:
- Looks like a duplicate of the https://github.com/ros2/rosbag2/issues/1914
- The RCA can be found here https://github.com/ros2/rosbag2/issues/1914#issuecomment-2669973810

Crola1702, 2025-05-22:
Closing in favor of #1914