ros2/rosbag2 #2022: Provide a full interface to the serialized message on python

EmmanuelMess, 2025-05-26:
The current read interface for rosbags in python is:
https://github.com/ros2/rosbag2/blob/aee5ae1b0a66fb2b11e72d5c9c39c5e01da16f52/rosbag2_py/src/rosbag2_py/_reader.cpp#L48-L56

But this is missing the send_timestamp.

Which is available to cpp:
https://github.com/ros2/rosbag2/blob/aee5ae1b0a66fb2b11e72d5c9c39c5e01da16f52/rosbag2_cpp/src/rosbag2_cpp/reader.cpp#L69-L72

MichaelOrlov, 2025-05-28:
Need to extend the Python API to return the `send_timestamp` in the tuple as well.  
However, help is needed with implementation and would be appreciated.

omsv135, 2025-06-05:
As far as I understand, we just need to add `send_timestamp` in the tuple, and modify the relevant test cases in this repository. Anything else that is required to resolve this issue? Any docs to be updated?

Also, this is pretty basic function, anyone using `rosbag2_py` via unpacking `reader.read_next()` will need to make trivial changes in a lot of places (similar to tests defined here). Should this be a concern?

EmmanuelMess, 2025-06-05:
> Also, this is pretty basic function, anyone using rosbag2_py via unpacking reader.read_next() will need to make trivial changes in a lot of places (similar to tests defined here). Should this be a concern?

This will break a lot of scripts, I would keep the original function, maybe mark it deprecated.

PR #2061 (2025-07-03): Add send_timestamp to python interface for reading serialized messages
## Description

Fixes #2022

The cpp interface for reading rosbags provides two timestamps, namely `recv_timestamp` and `send_timestamp`, from the `read_next` function. However, the python interface only returns `recv_timestamp`. This PR extends the Python API to return both timestamps.

Related changes:
- Python stub files are generated as mentioned in [this readme](https://github.com/ros2/rosbag2/blob/rolling/rosbag2_py/README.md).
- Relevant tests have also been updated in rosbag2_py.

### Is this user-facing behavior ch [... cut]