ros2/rosbag2 #2037: `rosbag2_performance_benchmarking` deb install empty

TannerGilbert, 2025-06-09:
## Description
When installing `rosbag2_performance_benchmarking` via apt: `sudo apt install ros-humble-rosbag2-performance-benchmarking` I get a empty package and therefore when I try to call `benchmark_launch.py` I get an error:

```
ros2 launch rosbag2_performance_benchmarking benchmark_launch.py benchmark:=`ros2 pkg prefix rosbag2_performance_benchmarking`/share/rosbag2_performance_benchmarking/config/benchmarks/test.yaml producers:=`ros2 pkg prefix rosbag2_performance_benchmarking`/share/rosbag2_performance_benchmarking/config/producers/mixed_110Mbs.yaml
file 'benchmark_launch.py' was not found in the share directory of package 'rosbag2_performance_benchmarking' which is at '/opt/ros/humble/share/rosbag2_performance_ben
```
This is probably due to the missing `DBUILD_ROSBAG2_BENCHMARKS=1` that is required when building as mentioned in https://github.com/ros2/rosbag2/tree/rolling/rosbag2_performance/rosbag2_performance_benchmarking#building

### Expected Behavior

`apt` install should work build the package similar to when running `colcon build --packages-select rosbag2_performance_benchmarking --cmake-args -DBUILD_ROSBAG2_BENCHMARKS=1` for source install.

## To Reproduce
1. `sudo apt install ros-humble-rosbag2-performance-benchmarking`
2. `ros2 launch rosbag2_performance_benchmarking benchmark_launch.py benchmark:=`ros2 pkg prefix rosbag2_performance_benchmarking`/share/rosbag2_performance_benchmarking/config/benchmarks/test.yaml producers:=`ros2 pkg prefix rosbag2_performance_benchmarking`/share/rosbag2_performance_benchmarking/config/producers/mixed_110Mbs.yaml`

## System
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble
 - Install Method: APT
 - Version: Humble

MichaelOrlov, 2025-07-11:
@yassine-cherni, Please consider handling this issue.

ivaughn-spear, 2025-07-11:
Fair enough.  As a clarification-- what's the reasoning we don't want to be running it on CI?  Combination of "lots of resources/$$" and "benchmarking your CI server doesn't help my robot"?  Maybe other reasons?

MichaelOrlov, 2025-07-12:
Update:
-  We were expecting to include https://github.com/ros2/rosbag2/pull/2073 in the upcoming Juzzy release 0.26.8.

However, multiple other issues in the `rosbag2_performance_benchmarking` package were discovered, so we decided to defer it for the next release, perhaps 0.26.9.

Please refer to the newly opened PRs:
 1. https://github.com/ros2/rosbag2/issues/2075
 2. https://github.com/ros2/rosbag2/pull/2077
 3. https://github.com/ros2/rosbag2/pull/2078

PR #2093 (2025-07-25): Enable `rosbag2_performance_benchmarking` package to be built by default
## Description

- This PR enables `rosbag2_performance_benchmarking` package to be built by default.

- Fixes #2037

### Is this user-facing behavior change?
No.

### Did you use Generative AI?
Yes, I used GitHub Copilot, Claude Sonnet 3.7 Thinking to help with writing comments and updates in the readme file.

### Additional Information
Note: the tests from the `rosbag2_performance_benchmarking` package are excluded from the build by default and should be explicitly enabled during the build time via ``--cmake-args -DENABL [... cut]
PR #2096 (2025-07-26): [kilted] Enable `rosbag2_performance_benchmarking` package to be built by default (backport #2093)
## Description

- This PR enables `rosbag2_performance_benchmarking` package to be built by default.

- Fixes #2037

### Is this user-facing behavior change?
No.

### Did you use Generative AI?
Yes, I used GitHub Copilot, Claude Sonnet 3.7 Thinking to help with writing comments and updates in the readme file.

### Additional Information
Note: the tests from the `rosbag2_performance_benchmarking` package are excluded from the build by default and should be explicitly enabled during the build time  [... cut]
PR #2097 (2025-07-26): [jazzy] Enable `rosbag2_performance_benchmarking` package to be built by default (backport #2093)
## Description

- This PR enables `rosbag2_performance_benchmarking` package to be built by default.

- Fixes #2037

### Is this user-facing behavior change?
No.

### Did you use Generative AI?
Yes, I used GitHub Copilot, Claude Sonnet 3.7 Thinking to help with writing comments and updates in the readme file.

### Additional Information
Note: the tests from the `rosbag2_performance_benchmarking` package are excluded from the build by default and should be explicitly enabled during the build time v [... cut]
PR #2098 (2025-08-07): [humble] Enable `rosbag2_performance_benchmarking` package to be built by default (backport #2093)
## Description

- This PR enables `rosbag2_performance_benchmarking` package to be built by default.

- Fixes #2037

### Is this user-facing behavior change?
No.

### Did you use Generative AI?
Yes, I used GitHub Copilot, Claude Sonnet 3.7 Thinking to help with writing comments and updates in the readme file.

### Additional Information
Note: the tests from the `rosbag2_performance_benchmarking` package are excluded from the build by default and should be explicitly enabled during the build time  [... cut]