ros2/rosbag2 #2094: rosbag2 playback causes TF_OLD_DATA and "trajectory ends in the past" errors

flupinette, 2025-07-24:
## Description
When replaying a bag recorded with ros2 bag record on a real robot with RViz2 and MoveIt2 (ROS 2 Jazzy), I get constant TF_OLD_DATA warnings or errors like "[ERROR] [xarm7_traj_controller]: Received trajectory with non-zero start time (1753350015.522340) that ends in the past (1753350015.702326)" (depending on whether i recorded \tf topics or not).

### Expected Behavior
The bag should replay smoothly. RViz2 should not complain about outdated transforms (TF_OLD_DATA).

### Actual Behavior
I get either this error :
```TF_OLD_DATA ignoring data from the past for frame left_finger at time 1753360039.580997
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained```

Or this one :

```[xarm7_traj_controller]: Received trajectory with non-zero start time (1753350015.522340) that ends in the past (1753350015.702326)```

No trajectory is executed.

## To Reproduce
1. Record a bag
2. Play it back
3. Observe the error

## System (please complete the following information)
 - OS: Ubuntu 24.04 (WSL2)
 - ROS 2 Distro: Jazzy
 - Install Method: APT
 - Version: 0.26.7

flupinette, 2025-07-25:
I work on a xArm7 and with this repository : https://github.com/xArm-Developer/xarm_ros2/tree/jazzy. 

I launch the connection with the robot with : `ros2 launch xarm_moveit_servo xarm_moveit_servo_realmove.launch.py robot_ip:=192.168.1.240 dof:=7 add_gripper:=True` and then after i run `ros2 run xarm_moveit_servo xarm_keyboard_input` to move the robot. I run `ros2 bag record -o demo \
  /xarm7_traj_controller/joint_trajectory \
  /xarm/joint_states \
  /xarm/robot_states \
  /tf \
  /tf_static` and move the robot. When it's done, I stop the recording with ^C and then i use `ros2 bag play demo`. Nothing moves and in the logs of the `ros2 launch xarm_moveit_servo xarm_moveit_servo_realmove.launch.py robot_ip:=192.168.1.240 dof:=7 add_gripper:=True` I get errors like : 
```
[rviz2-1] [WARN] [1753433223.572722489] []: TF_OLD_DATA ignoring data from the past for frame link2 at time 1753433204.038107 according to authority Authority undetectable
[rviz2-1] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[rviz2-1] [WARN] [1753433223.572729045] []: TF_OLD_DATA ignoring data from the past for frame link3 at time 1753433204.038107 according to authority Authority undetectable
[rviz2-1] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[rviz2-1] [WARN] [1753433223.572733860] []: TF_OLD_DATA ignoring data from the past for frame link4 at time 1753433204.038107 according to authority Authority undetectable
[rviz2-1] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[rviz2-1] [WARN] [1753433223.572739596] []: TF_OLD_DATA ignoring data from the past for frame link5 at time 1753433204.038107 according to authority Authority undetectable
[rviz2-1] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[rviz2-1] [WARN] [1753433223.572747995] []: TF_OLD_DATA ignoring data from the past for frame link6 at time 1753433204.038107 according to authority Authority undetectable
[rviz2-1] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[rviz2-1] [WARN] [1753433223.572696164] []: TF_OLD_DATA ignoring data from the past for frame link4 at time 1753433204.038107 according to authority Authority undetectable
[rviz2-1] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[rviz2-1] [WARN] [1753433223.572779033] []: TF_OLD_DATA ignoring data from the past for frame link5 at time 1753433204.038107 according to authority Authority undetectable
[rviz2-1] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[rviz2-1] [WARN] [1753433223.572753834] []: TF_OLD_DATA ignoring data from the past for frame link7 at time 1753433204.038107 according to authority Authority undetectable
[rviz2-1] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[rviz2-1] [WARN] [1753433223.572827996] []: TF_OLD_DATA ignoring data from the past for frame left_finger at time 1753433204.038107 according to authority Authority undetectable
[rviz2-1] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained```

If I record with just `ros2 bag record -o demo \
  /xarm7_traj_controller/joint_trajectory \
  /xarm/joint_states \
  /xarm/robot_states` and then I replay, I get :

```
[ros2_control_node-3] [ERROR] [1753350023.636827401] [xarm7_traj_controller]: Received trajectory with non-zero start time (1753350015.522340) that ends in the past (1753350015.702326)
[ros2_control_node-3] [ERROR] [1753350023.646968038] [xarm7_traj_controller]: Received trajectory with non-zero start time (1753350015.522340) that ends in the past (1753350015.712325)
[ros2_control_node-3] [ERROR] [1753350023.656750625] [xarm7_traj_controller]: Received trajectory with non-zero start time (1753350015.532319) that ends in the past (1753350015.722314)
[ros2_control_node-3] [ERROR] [1753350023.666798925] [xarm7_traj_controller]: Received trajectory with non-zero start time (1753350015.552346) that ends in the past (1753350015.732300)
[ros2_control_node-3] [ERROR] [1753350023.676902939] [xarm7_traj_controller]: Received trajectory with non-zero start time (1753350015.552346) that ends in the past (1753350015.742337)
[ros2_control_node-3] [ERROR] [1753350023.687100654] [xarm7_traj_controller]: Received trajectory with non-zero start time (1753350015.562352) that ends in the past (1753350015.752305)
[ros2_control_node-3] [ERROR] [1753350023.696801715] [xarm7_traj_controller]: Received trajectory with non-zero start time (1753350015.572348) that ends in the past (1753350015.762348)
[ros2_control_node-3] [ERROR] [1753350023.706788457] [xarm7_traj_controller]: Received trajectory with non-zero start time (1753350015.582338) that ends in the past (1753350015.772331)
[ros2_control_node-3] [ERROR] [1753350023.716765750] [xarm7_traj_controller]: Received trajectory with non-zero start time (1753350015.592356) that ends in the past (1753350015.782335)```

MichaelOrlov, 2025-07-25:
@flupinette Your description sounds like a more generic problem on some concrete system and setup, which needs to be triaged and RCA (Root Cause Analysis) needs to be done.  
Then, when the problem will be isolated and if it belongs to the Rosbag2, we can consider how to fix it.

cc: @emersonknapp may maybe you have some thoughts about this problem or came across a similar issue in your practice?

closed 2026-05-10 by flupinette