ros2/rosbag2 #2110: get_typesupport_library Does Not Validate Invalid Type Strings

zhihaoshang, 2025-08-04:
## Description
The function get_typesupport_library(const std::string & type, const std::string & typesupport_identifier) does not perform validation on malformed or illegal type strings.

### Expected Behavior
throw a std::runtime_error

### Actual Behavior
without throwing any exception

## To Reproduce
### Test Case
```cpp
#include <gmock/gmock.h>
#include <memory>
#include <string>
#include <utility>
#include "ament_index_cpp/get_package_prefix.hpp"
#include "rcpputils/shared_library.hpp"
#include "rosbag2_cpp/typesupport_helpers.hpp"
using namespace ::testing;  // NOLINT

TEST(TypesupportHelpersTest, throws_exception_with_special_characters_in_type) {
  EXPECT_THROW(
    rosbag2_cpp::get_typesupport_library("test_msgs/!@#$%", "rosidl_typesupport_cpp"),
    std::runtime_error);
}
```

### Output
```markdown
[ RUN      ] TypesupportHelpersTest.throws_exception_with_special_characters_in_type
/home/shangzh/rosbag2_ws/rosbag2/rosbag2_cpp/test/rosbag2_cpp/test_typesupport_helpers.cpp:11: Failure
Expected: rosbag2_cpp::get_typesupport_library("test_msgs/!@#$%", "rosidl_typesupport_cpp") throws an exception of type std::runtime_error.
  Actual: it throws nothing.

[  FAILED  ] TypesupportHelpersTest.throws_exception_with_special_characters_in_type (3 ms)
[----------] 1 test from TypesupportHelpersTest (3 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (3 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TypesupportHelpersTest.throws_exception_with_special_characters_in_type

 1 FAILED TEST
```

## System (please complete the following information)
OS: ubuntu 24.04
ROS 2 Distro: ros 2 jazzy
Install Method: source
Version: ros 2 jazzy

## Additional context
build options: --mixin asan-gcc

Barry-Xu-2018, 2025-08-04:
The invalid characters "!@#$%" do not affect the extraction of the package name.

https://github.com/ros2/rosbag2/blob/26717b1eb6c3f5995ac28ca578d59cc834b67584/rosbag2_cpp/src/rosbag2_cpp/typesupport_helpers.cpp#L76-L99

It successfully extracted the package name "test_msgs" and found the related library. So no error occurred when calling rosbag2_cpp::get_typesupport_library().   

"!@#$%" will affect `rosbag2_cpp::get_typesupport_handle()`.

MichaelOrlov, 2025-09-22:
- Closing as it wouldn't be fixed in Rosbag2 since all typesupport utility functions were moved to the `rclcpp` package. Please refer to https://github.com/ros2/rosbag2/pull/2017 for details.  
- Feel free to open a similar issue in the [rclcpp](https://github.com/ros2/rclcpp) repo.

closed 2025-09-22 by MichaelOrlov